Gasman2014/KiCad-Diff #5: TypeError: 'Variable' object is not subscriptable

metropt, 2019-03-28:
Hi,

I tried to run it but I always get the following error:

```
Git
Resolution (dpi) :  300
Commit1 PY_VAR22
Commit2 PY_VAR23
Traceback (most recent call last):
  File "kidiff_gui.py", line 1103, in <module>
    times = getGitDiff(d1, d2, prjctName, prjctPath)
  File "kidiff_gui.py", line 72, in getGitDiff
    artifact1 = diff1[:6]
TypeError: 'Variable' object is not subscriptable
```

I notice that sometimes I got my commit hash instead of PY_VAR22/23 but I can't understand why since I always seletec two commits.

I'm running it with Python3.5

Thank you

Gasman2014, 2019-03-29:
I will have a look at this but I am working on a new and MUCH faster version which I hope to get out in the next week or two so I can’t give this much attention at the moment. Quick check, Have you also installed the plotpcb.py script from the repo in the correct place? Can you confirm that your paths are correct for the git program etc as set in the first few lines of the program. 

John Pateman

tonio976, 2019-08-25:
It happens to me when i only check one ''commit'' over the two expected for the comparaison..

Best regards, 
Antoine

Gasman2014, 2019-08-25:
What SCM are you using? This script can work with Git, SVN and Fossil but the processing of the artefact names is different for each.

JP

tonio976, 2019-08-26:
i'm using git on mac.

Gasman2014, 2019-08-26:
Ok, will have a look at that - I think I have an idea about why this is not working but need some time to test. In the meanwhile, does providing two references work ok? Are you using the newer, faster branch?  I added Git as it seems it is used more than other SCMs but it is the least tested and I have less experience with Git so I’m not so surprised. 

If you are not totally committed to Git, I would encourage you to take a look at Fossil which is very well suited to EDA work. http://www.fossil-scm.org/home/doc/trunk/www/index.wiki

John Pateman

tonio976, 2019-08-26:
Lol, i cannot cannot manage to get my diff in a png (i get empty folders) but i'm not sure 100% of my installation.

Gasman2014, 2019-09-01:
Are you using the Python gui script (kidiff_gui.py) or the old shell script (kidiff.sh). The latter should really be removed as I don’t plan to develop it any further and don’t have time to support it.

I am presently only developing the kidiff_gui.py on the  ‘SVG’ branch - so I suggest that you switch to this branch - it is MUCH  faster.  Does the initial window of commits appear?

Gasman2014, 2020-03-02:
New version developed and am no longer working on either of the older versions.

closed 2020-03-02 by Gasman2014