Gasman2014/KiCad-Diff #6: feature request: covering for Eeschema diff

easyw, 2019-05-15:
As in the title, would it possible to cover also Eeschema with your diff tools?
That would be a great addition 😸

Gasman2014, 2019-05-15:
The automated generation of the reference files to compare is done via the PcbNew Python interface. The current version compares two png files generated programatically but it turns out that this was not a good design choice as the processing is very, very slow. A new version that I now have working uses a comparison method based on SVG files which is WAY faster. (However, I really need a hand from a JS/HTML expert to finally sort out a couple of display issues)

Unfortunately,  there is not a method currently to programatically generate any form of files directly from Eeschema. To generate the images myself from the .sch file would be quite a lot of work but this is the route taken by these two projects which might meet your needs.

http://neo900.org/stuff/eeshow/ <http://neo900.org/stuff/eeshow/>
https://github.com/jnavila/plotkicadsch <https://github.com/jnavila/plotkicadsch>

FYI The new Diff looks a bit like this 

JP

easyw, 2019-05-15:
Hi John @Gasman2014
Thanks for the tips... plotkicadsch seems can cover eeschema diffs...
BTW I woul like to test your KiCAD-Diff with svgs...
PS the image you linked is not visible to me...

Gasman2014, 2019-05-15:
I will push my current code to a branch when i get a chance - I need to do a bit of surgery to the interface at some point but you can still use it with the old interface but just ignore the layers and resolution settings.
Not sure why the image is not visible - I have changed a couple of them to pdfs and have tried again.

easyw, 2019-05-15:
PS still not visible to me...
anyway I'm going to close it and wait for your svg branch :smile: 
Thx

closed 2019-05-15 by easyw

Gasman2014, 2019-05-16:
I forgot to mention that there is this ugly hack way of generating SVGs form eeschema using a virtual frame buffer to simulate a display and passing hardcoded keyboard shortcuts.  

http://scottbezek.blogspot.com/2016/04/automated-kicad-schematic-export.html