Gasman2014/KiCad-Diff #7: plotPCB return empty

metropt, 2020-03-16:
Hi,

I tried to run the updated source files however it failed on get data from plotPCB.
```
Commit1 d07d4f3 	 Improved batt tabs pads;
Commit2 35b0361 	 Minor improvement
2020-03-16 15:51:50 2020-03-16 14:30:19
Generating .svg files
#THIS IS A PRINT OF THE COMMAND
/usr/local/bin/plotPCB.py /home/ze/development/pcb/Plots/d07d4f/pcb.kicad_pcb /home/ze/development/pcb/Plots/d07d4f [] []
Traceback (most recent call last):
  File "../KiCad-Diff/kidiff_gui.py", line 1549, in <module>
    svgDir1, svgDir2, boardDims1, boardDims2 = makeSVG(d1, d2, prjctName, prjctPath)
  File "../KiCad-Diff/kidiff_gui.py", line 1140, in makeSVG
    return (d1, d2, plotDims1[0], plotDims2[0])
IndexError: list index out of range

```

Running plotPCB
```
/usr/local/bin/plotPCB.py /home/ze/development/pcb/Plots/d07d4f/pcb.kicad_pcb /home/ze/development/pcb/Plots/d07d4f 
Traceback (most recent call last):
  File "/usr/local/bin/plotPCB.py", line 16, in <module>
    file = open('/tmp/svg/layers', 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/svg/layers'
```
 it failed to read a layers file however looking into Kicaddiff_gui.py I can't see where this file is created.

Thank you

Gasman2014, 2020-03-16:
I think I can see what I did there. The original code was v slow so I added in a facility to select only a subset of the layers. I thought I had stripped all the references to this out in the new code out but had left some directories in place which made it run ok on my system. I will have a look later this evening.

Gasman2014, 2020-03-16:
I have pushed a fix. Please try again with the new plotPCB.py file
 4adfe93

Gasman2014, 2020-03-20:
Could you confirm if the fix that I pushed has corrected the problem?  If so, I would like to close the issue.

metropt, 2020-03-20:
I change this line https://github.com/Gasman2014/KiCad-Diff/commit/4adfe93fecc55c7661e209e655b9e57075cd1b70#r37948454 and installed the new plotPCB.py and now it runs but I only get diff from the project information.

[image]

metropt, 2020-03-20:
BTW I'm using Ubuntu but I'll test it on OSX too.

Gasman2014, 2020-03-21:
Can you post the complete output from the terminal.

metropt, 2020-03-21:
```ze@ze-XPS-13-9380:~/development/pcb$ python3 ~/development/KiCad-Diff/kidiff_gui.py 
:0
Commit1 35b0361 	 Minor improvement
Commit2 62b564c 	 Finished Layout
2020-03-16 14:30:19 2020-03-12 11:45:31
Generating .svg files
196502286 63702500 35036049 29440301 196502286 63702500 35036049 29440301
{'title': '', 'rev': '', 'company': '', 'date': '', 'page': 'A4 ', 'thickness': '0.8 ', 'drawings': '40 ', 'tracks': '244 ', 'zones': '0 ', 'modules': '27 ', 'nets': '29 '}
{'title': '', 'rev': '', 'company': '', 'date': '', 'page': 'A4 ', 'thickness': '0.8 ', 'drawings': '41 ', 'tracks': '257 ', 'zones': '0 ', 'modules': '27 ', 'nets': '29 '}
serving at port 9090
127.0.0.1 - - [21/Mar/2020 18:07:52] "GET /web/index.html HTTP/1.1" 200 -
127.0.0.1 - - [21/Mar/2020 18:07:52] "GET /web/style.css HTTP/1.1" 200 -
```

metropt, 2020-03-21:
```
ze@ze:~/development/pcb$ cd Plots/
ze@ze:~/development/pcb/Plots$ ls -la
total 52
drwxr-xr-x 13 ze ze 4096 mar 21 17:42 .
drwxr-xr-x 15 ze ze 4096 mar 21 17:44 ..
drwxr-xr-x  2 ze ze 4096 mar 20 18:50 270363
drwxr-xr-x  2 ze ze 4096 mar 20 18:50 35b036
drwxr-xr-x  2 ze ze 4096 mar 20 18:44 560546
drwxr-xr-x  2 ze ze 4096 mar 20 18:31 56e592
drwxr-xr-x  2 ze ze 4096 mar 20 18:50 62b564
drwxr-xr-x  2 ze ze 4096 mar 20 18:50 6a6860
drwxr-xr-x  2 ze ze 4096 mar 21 17:42 6ece64
drwxr-xr-x  2 ze ze 4096 mar 20 18:31 6f381a
drwxr-xr-x  2 ze ze 4096 mar 20 18:37 8daa9c
drwxr-xr-x  2 ze ze 4096 mar 20 18:39 ee6514
drwxr-xr-x  3 ze ze 4096 mar 20 18:50 web
ze@ze:~/development/pcb/Plots$ cd 6ece64/
ze@ze:~/development/pcb/Plots/6ece64$ ls
cointag.kicad_pcb
ze@ze:~/development/pcb/Plots/6ece64$ 
```

Gasman2014, 2020-03-23:
Quick check -  Please can you double check that all the expected programs listed in lines 39-47 in the main app are where they are expected to be or set to '' ? The first line of the log should list the  VCS programs found on your machine. i.e. Should say Fossil, Git, SVN but yours says :0 which isn't right.

Also, have you renamed the layers or are they in a non-English language? I have made an assumption that they are called F_Cu, B_Mask etc. I am not sure how they are set if the interface is non-English. I have pushed a further commit with some extra debugging. I have been running this with no problems on a number of git archives with no problems. The new debugging prints out the actual command that is issued to see if the files differ. 
BTW You will only see the name of the board etc if you enter it in the 'File' > 'Page Settings' menu.

a-allard, 2020-04-26:
Hi I was having the same issue.
I fixed it by importing pcbnew at the top of the plotPCB.py file.  It seems it should not be
```python
from pcbnew import *
```
but it should be
```python
from pcbnew import *
import pcbnew
```

Or line 15 (I think 14 in the original file) should be `pcbnew.LoadBoard(boardName)`

Also another issue I came across was in kidiff_gui.py
```python
    gui = tk.Tk(SCMS)
```
I had to make:
```python
    gui = tk.Tk(':0.0', SCMS)
```

For reference I am running on Ubuntu 20.04 with Kicad 5.1.5

All in all though this is an Awesome Tool!!! It has saved me from having to restore a board to a different commit to see what has changed.

If you'd like I can push up what I have to a different branch and create an MR.  Let me know.

Gasman2014, 2020-04-27:
Thanks for testing this - I have only really checked it out on macOS so some silly things (like the missing import) have probably slipped through. I have made the edits without a PR as they are so small. The gui directive doesn't seem to be necessary on macOS but it seems harmless so I have pushed it to master.

BenRoe, 2020-08-30:
I get the following error on `OSX 10.14.6` and `Kicad 5.1.6`

```bash
▶python kidiff_gui.py
Fossil
Git
SVN
2020-08-30 19:59:34.266 python[50129:3401772] [QL] file:///usr/local/Caskroom/qlmarkdown/1.3.4/QLMarkdown.qlgenerator/ has no identifier: ignoring
2020-08-30 19:59:34.279 python[50129:3401772] [QL] Can't get plugin bundle info at file:///Users/Ben/Library/Application%20Support/Autodesk/webdeploy/production/9734d4c0ebc0481cf97bf881c8c462446cf2c899/Autodesk%20Fusion%20360.app/Contents/Library/QuickLook/NQLGenerator.qlgenerator
2020-08-30 19:59:34.279 python[50129:3401772] [QL] Can't get plugin bundle info at file:///Users/Ben/Library/Application%20Support/Autodesk/webdeploy/production/ea8c8acaa650cc6e8cc494435c12dd5f2e5c8b5f/Autodesk%20Fusion%20360.app/Contents/Library/QuickLook/NQLGenerator.qlgenerator
2020-08-30 19:59:34.281 python[50129:3401772] [QL] Can't get plugin bundle info at file:///Users/Ben/Library/Application%20Support/Autodesk/webdeploy/production/bdb38befef4f9ecb49238115faae42904b54fa67/Autodesk%20Fusion%20360.app/Contents/Library/QuickLook/NQLGenerator.qlgenerator
objc[50129]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fff905793d8) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x113008f50). One of the two will be used. Which one is undefined.
Commit1 624a621 	 fixed some labels and modified mini-e cutout
Commit2 bcc667b 	 rev1.1
cd /Users/Ben/Git/Lily58L/pcb/PCB && /usr/bin/git diff --name-only 624a62 bcc667 | /usr/bin/grep *.kicad_pcb
cd /Users/Ben/Git/Lily58L/pcb/PCB && /usr/bin/git show 624a62: > /Users/Ben/Git/Lily58L/pcb/PCB/Plots/624a62/Lily58_Pro.kicad_pcb cd /Users/Ben/Git/Lily58L/pcb/PCB && /usr/bin/git show bcc667: > /Users/Ben/Git/Lily58L/pcb/PCB/Plots/bcc667/Lily58_Pro.kicad_pcb
cd /Users/Ben/Git/Lily58L/pcb/PCB && /usr/bin/git show -s --format="%ci" 624a62 cd /Users/Ben/Git/Lily58L/pcb/PCB && /usr/bin/git show -s --format="%ci" bcc667
2020-08-30 16:14:25 2020-08-08 16:10:39
Generating .svg files
/usr/local/bin/plotPCB.py /Users/Ben/Git/Lily58L/pcb/PCB/Plots/624a62/Lily58_Pro.kicad_pcb /Users/Ben/Git/Lily58L/pcb/PCB/Plots/624a62 /usr/local/bin/plotPCB.py /Users/Ben/Git/Lily58L/pcb/PCB/Plots/bcc667/Lily58_Pro.kicad_pcb /Users/Ben/Git/Lily58L/pcb/PCB/Plots/bcc667
[] []
Traceback (most recent call last):
  File "kidiff_gui.py", line 1591, in <module>
    svgDir1, svgDir2, boardDims1, boardDims2 = makeSVG(d1, d2, prjctName, prjctPath)
  File "kidiff_gui.py", line 1182, in makeSVG
    return (d1, d2, plotDims1[0], plotDims2[0])
IndexError: list index out of range
```

Gasman2014, 2020-12-31:
Sorry taken so long to look at this.

Have you given the file `plotPCB_macOS.py` or `plotPCB.py` executable permissions? 
You need to do a `chmod a+x plotPCB_macOS.py` to allow it to plot the actual diffs.

purple-bobby, 2021-02-07:
Like BenRoe, I also have problem with pcbnew
````
...
Plot 1 Command    :  plotpcb /home/bobby/hw/kidiff/1203ca/dopp-link.kicad_pcb -o /home/bobby/hw/kidiff/1203ca
Plot 2 Command    :  plotpcb /home/bobby/hw/kidiff/5c6604/dopp-link.kicad_pcb -o /home/bobby/hw/kidiff/5c6604

Plot1 error: Traceback (most recent call last):
  File "/home/bobby/KiCad-Diff/plotPCB.py", line 16, in <module>
    import pcbnew
  File "/usr/lib64/python/site-packages/pcbnew.py", line 38, in <module>
    import _pcbnew
ImportError: dynamic module does not define module export function (PyInit__pcbnew)
````

I'm using openSUSE Leap 15.2 and KDE Plasma 5.18.6 on i5
````
bobby@linux-4rsh:~/hw> uname -a
Linux linux-4rsh 5.3.18-lp152.63-default #1 SMP Mon Feb 1 17:31:55 UTC 2021 (98caa86) x86_64 x86_64 x86_64 GNU/Linux
````
I notice `/usr/lib64/python3.6/site-packages/` does not have `pcbnew.py`

pcbnew is in
````
/usr/lib64/python2.7/site-packages/_pcbnew.so
/usr/lib64/python2.7/site-packages/pcbnew.py
````

leoheck, 2021-06-18:
@metropt this is a pretty old issue. Could you check if it can be closed? Thank you.

Gasman2014, 2021-06-28:
Think this should be fixed so I'm closing this. Please reopen if there are still issues with the latest version.

closed 2021-06-28 by Gasman2014