Gasman2014/KiCad-Diff #8: Does not handle filenames with spaces on Linux

sunwukonga, 2020-05-06:
I haven't tested on a Windows or MacOS system.
To reproduce:
* `git clone https://github.com/Ergodox-io/ErgoDox`
* run `kidiff_gui.py` and navigate to `ErgoDOX/ErgoDOX\ pcb/ErgoDOX.kicad_pcb`

I did get it working, but only after substantial modifications that I cannot guarantee would work on either Windows or MacOS.

I used the following to escape each instance of the path when used in a process call.
```
if sys.version_info[0] >= 3:
  unicode = str
 
def _escape_string( val ):
  # Make unicode
  val = unicode( val )
  # Escape stuff
  val = val.replace( u'\\', u'\\\\' )
  val = val.replace( u' ', u'\\ ' )
  return val
```

I also did away with many references to the path by using the linux /tmp directory instead of putting the generated files in the repo itself.

Gasman2014, 2020-05-06:
Previous plot generation was quite slow so I chose to plot the results within the project directory itself and check for the presence of an already rendered version prior to avoid regenerating a plot.

I always avoid spaces in project and path names so this has not bitten me. It would be a helpful improvement but I don't have a great deal of time at present. Could you make a pull request so I can check that any alteration still works in macOS?

sunwukonga, 2020-05-07:
I avoid spaces myself. I inherited the file and directory names in the instance that tripped me up.

I'll try to put together a PR soon, weather and time permitting :)

sunwukonga, 2020-05-09:
Leaving open for a future where spaces are handled properly in Windows.

leoheck, 2021-06-18:
Hi @sunwukonga can you check if this issue still stands? Otherwise, it will be nice to close it. Cool?

jnavila, 2021-06-18:
To me, it should have been fixed with the conversion of SCM commands to full python exec calls.

leoheck, 2021-06-18:
So, it was not fixed then. I also tested it here by renaming a repo using spaces in it and it is breaking because of that. Do we really need this? Who uses spaces in filenames anyways?

jnavila, 2021-06-18:
Let me check how it fails. There's surely a real bug behind.

leoheck, 2021-06-18:
This is how it is failing.
```
~/Documents 
➜ mv assoc-board "assoc board"

~/Documents 
➜ cd assoc\ board 

assoc board on  main [?] 
➜ kidiff board.kicad_pcb 
Traceback (most recent call last):
  File "/home/lheck/Dropbox/Documents/KiCad-Diff_lheck/kidiff", line 604, in <module>
    prjctPath, kicad_project = scm.get_kicad_project_path(settings.escape_string(kicad_project_path))
  File "/home/lheck/Dropbox/Documents/KiCad-Diff_lheck/scms/git.py", line 100, in get_kicad_project_path
    stdout, _ = settings.run_cmd(prjctPath, cmd)
  File "/home/lheck/Dropbox/Documents/KiCad-Diff_lheck/settings.py", line 46, in run_cmd
    p = Popen(
  File "/usr/lib/python3.9/subprocess.py", line 951, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.9/subprocess.py", line 1821, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/lheck/Dropbox/Documents/assoc\\ board'
```

leoheck, 2021-06-18:
I see. The double "\\" is the issue here.

It is ok to stack multiples folder separators they get ignored
`/`, `//`, `///`, `/////////////` <= these are all the same.

However, these 2 are not the same `\ ` `\\ `

leoheck, 2021-06-18:
Now I understod why this function exists. And it looks like a not-good workaround.
```
def escape_string(val):

    if sys.version_info[0] >= 3:
        unicode = str

    val = unicode(val)
    val = val.replace( u'\\', u'\\\\' )
    val = val.replace( u' ', u'\\ ' )

    return ''.join(val.splitlines())
```

jnavila, 2021-06-18:
You no longer need to escape spaces in paths with my refactor.

leoheck, 2021-06-18:
\o/ I was struggling a lot to fix this. Let me check then.

leoheck, 2021-06-18:
Where your refactor is?

jnavila, 2021-06-18:
Can you try #58?

leoheck, 2021-06-18:
It was not working fully. I made a PR with a fix in your fork. https://github.com/jnavila/KiCad-Diff/pull/1

jnavila, 2021-06-18:
Please open a PR directly here. It's more direct and safer.

leoheck, 2021-06-18:
Cool. I was thinking that since you was fixing this it was faster to have this included in your PR.

leoheck, 2021-06-20:
Hey @sunwukonga this issue was addressed and the code  that fix it was already merged. Could you check if you can close this issue? If the issue persists it can be reopened in future too.

Gasman2014, 2021-06-28:
This has been fixed and recent testing, on macOS at least, confirms this. Closing.

closed 2021-06-28 by Gasman2014