Gasman2014/KiCad-Diff #9: Fails if the `.kicad_pcb` file is in a subdirectory.

sunwukonga, 2020-05-06:
This is caused by line [700](https://github.com/Gasman2014/KiCad-Diff/blob/1497b87bce3ca9425062dee4a6082b0b6e6f1267/kidiff_gui.py#L700) of the `kidiff_gui.py` file.

I solved for myself by hard-coding my subdirectory name in place, but I'm sure there's a much better approach :)

Gasman2014, 2020-05-06:
Needs to include better path handling and accommodate alternative file locations.

sunwukonga, 2020-05-07:
`"git ls-tree -r --name-only HEAD | grep -m 1 " + prjctName`

Will give the correct git filepath. I.e. the following should work:
```
gitPath = "git ls-tree -r --name-only HEAD | grep -m 1 " + prjctName

gitArtifact1 = 'cd ' + _escape_string(prjctPath) + ' && ' + gitProg + ' show ' + artifact1 + \
      ':' + gitPath + ' > ' + outputDir1 + '/' + prjctName
gitArtifact2 = 'cd ' + _escape_string(prjctPath) + ' && ' + gitProg + ' show ' + artifact2 + \
      ':' + gitPath + ' > ' + outputDir2 + '/' + prjctName
```

Where `_escape_string` is the function described on the spaces in paths [issue](https://github.com/Gasman2014/KiCad-Diff/issues/8#issue-613034458).

closed 2020-05-09 by sunwukonga