Gasman2014/KiCad-Diff #11: I need help to run this for the 1st time

leoheck, 2020-05-27:
Hi, do you know what is going on here? 

```
➜ /home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py -h
Commit1 e80aa7a 	 Update step model for the 0.3.2 version.
Commit2 3454d1a 	 Fix links.
cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show -s --format="%ci" e80aa7 cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show -s --format="%ci" 3454d1
2020-05-25 17:09:56 2020-04-30 15:44:47
Generating .svg files
Traceback (most recent call last):
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py", line 1594, in <module>
    svgDir1, svgDir2, boardDims1, boardDims2 = makeSVG(d1, d2, prjctName, prjctPath)
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py", line 1184, in makeSVG
    return (d1, d2, plotDims1[0], plotDims2[0])
IndexError: list index out of range
```

leoheck, 2020-05-27:
Running the gui I got this
I fixed it by replacing the `_escape_string(gitRoot)` with `gitRoot` now
```
➜ /home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_gui.py
Fossil 
Git 
SVN 
Commit1 e80aa7a 	 Update step model for the 0.3.2 version.
Commit2 43f6d0f 	 More space for connector wires/strain relief.
cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git diff --name-only e80aa7 43f6d0 | /usr/bin/grep *.kicad_pcb
Traceback (most recent call last):
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_gui.py", line 1580, in <module>
    times = getGitDiff(d1, d2, prjctName, prjctPath)
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_gui.py", line 727, in getGitDiff
    gitPath = getGitPath(prjctName, prjctPath)
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_gui.py", line 676, in getGitPath
    gitPathCmd = 'cd ' + _escape_string(gitRoot) + ' && ' + gitProg + ' ls-tree -r --name-only HEAD | ' + grepProg + ' -m 1 ' + prjctName
NameError: name '_escape_string' is not defined
```

leoheck, 2020-05-27:
I found that some commands are not good. At least the output in the console.
Some `cd` commands do not have a comma before. So these commands look strange to me

cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && \
	/usr/bin/git diff --name-only e80aa7 3454d1 | /usr/bin/grep *.kicad_pcb

cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && \
	/usr/bin/git show e80aa7: > /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb/Plots/e80aa7/yardbird.kicad_pcb cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show 3454d1: > /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb/Plots/3454d1/yardbird.kicad_pcb

cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && \
	/usr/bin/git show -s --format="%ci" e80aa7 cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show -s --format="%ci" 3454d1

sunwukonga, 2020-05-27:
> Running the gui I got this
> I fixed it by replacing the `_escape_string(gitRoot)` with `gitRoot` now
>
> ```
> ➜ /home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_gui.py
> Fossil
> Git
> SVN
> Commit1 [...]

Can you try removing line 39:
```
val = val.replace( u'\\', u'\\\\' )
```

`_escape_string` is only there to escape spaces in file paths and names.

sunwukonga, 2020-05-27:
> I found that some commands are not good. At least the output in the console.
> Some cd commands do not have a comma before. So these commands look strange to me

What is not good, exactly? Why would a `cd` command need a `,` in front of it?

leoheck, 2020-05-27:
> What is not good, exactly? Why would a cd command need a , in front of it?

I don't know actually.

I am running your _gui.py 
I put the plotPCB.py to the path where your readme said to.

But the diff is not being generated. Do you have any ideia what should I do?

leoheck, 2020-05-27:
> First, what OS are you using?
Didn't see this.
Ubuntu 20.04

leoheck, 2020-05-27:
The _linux.py command line to generate images work.
The _gui works (or almost work) 
The Plot folder is generated
[image]

leoheck, 2020-05-27:
The content is this
[image]

sunwukonga, 2020-05-27:
You're using `kidiff_gui.py` OR `kidiff_linux.py` they are independent. If using `kidiff_linux.py` that's all you need.

leoheck, 2020-05-27:
I am trying both, actually. I would like to make the diff between git versions. The kidiff_linux.py generates only a folder with a bunch of svg's I didn't pass any commit to compare. Can I pass the git commit to that command? I didn't find it on REAMDE.

leoheck, 2020-05-27:
Do you have ane usage example? Maybe this is my problem. Maybe I am not using it right.

sunwukonga, 2020-05-27:
Can you post your repo? I can have a look from my end. I'll be able to diagnose more quickly.

sunwukonga, 2020-05-27:
The kidiff_gui.py will NOT be able to handle `.kicad_pcb` files that are not in the git root.

leoheck, 2020-05-27:
Yeah, that is what I was thinking about. Any reason for that? Or it is just how it was done now?

leoheck, 2020-05-27:
I am sorry, I can't share the repo.
But this is the layout of the folders
[image]

sunwukonga, 2020-05-27:
Just run `kidiff_linux.py` from anywhere is the usage. No need for switches or paths.

leoheck, 2020-05-27:
Yeah, I am doing this. It generates images of the board, right?

leoheck, 2020-05-27:
So, if this can't handle the board in a folder, I can still generate images with the kidiff_linux.py then use the imagemagic to compare them.

sunwukonga, 2020-05-27:
In the commit hash subdirectories of the `plots` directory, there should be many `svg` files generated. Do you see this?

leoheck, 2020-05-27:
If I can add here, I would like to suggest.

- Handling boards on different folders.
- Improving the command line.

I would suggest 

kidiff_gui.py

closed 2020-05-27 by leoheck

sunwukonga, 2020-05-27:
Are you then viewing the generated HTML at `http://127.0.0.1:9092/web/index.html`

leoheck, 2020-05-27:
If I can add here, I would like to suggest.

- Handling boards on different folders.
- Improving the command line.

I would suggest at least be able to pass the pcb from the command line. Then figure it out the main scm folder, and then the board folder inside the repo. Also if the user can pass the commit from the command line would be greate.

```
kidiff_gui.py <kicad_pcb> [-a commit] [-b commit]
```

leoheck, 2020-05-27:
> Are you then viewing the generated HTML at http://127.0.0.1:9092/web/index.html

 I am not even closer since my repo does not have the board on the root.

sunwukonga, 2020-05-27:
Thanks. That `_escape_string` should NOT be in `kidiff_gui.py` at all.

sunwukonga, 2020-05-27:
`kidiff_linux.py` can handle `kicad_pcb` files that aren't in root. There must be another cause.

leoheck, 2020-05-27:
Look this (using the 

```
Commit1 e80aa7a 	 Update step model for the 0.3.2 version.
Commit2 3454d1a 	 Fix links.
cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show -s --format="%ci" e80aa7 cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show -s --format="%ci" 3454d1
2020-05-25 17:09:56 2020-04-30 15:44:47
Generating .svg files
Traceback (most recent call last):
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py", line 1594, in <module>
    svgDir1, svgDir2, boardDims1, boardDims2 = makeSVG(d1, d2, prjctName, prjctPath)
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py", line 1184, in makeSVG
    return (d1, d2, plotDims1[0], plotDims2[0])
IndexError: list index out of range
```

This is the 3rd line on this log. I am braking the lines afer the && and adding a \ to be able to execute on the command line.
Look the cd commands, they are not good. Do you agree?

```
cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && \
	/usr/bin/git show -s --format="%ci" e80aa7 cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && \
	/usr/bin/git show -s --format="%ci" 3454d1

```

sunwukonga, 2020-05-27:
That `cd` command seems fine to me. I see your confusion though. You can ignore this. This is just a console print of the commands one after the other. It is NOT executing them there. 

The error is actually from a previous error. I.e. your `plotDims1` is not being generated properly.

leoheck, 2020-05-27:
Yes, true. Any idea on how I debug or how I can give you more info?

leoheck, 2020-05-27:
Just one FYI, here in ubuntu 20.04 I had to change this (kidiff_gui.py)
```
-    gui = tk.Tk(':0.0', SCMS)
+    gui = tk.Tk(':1.0', SCMS)
```

sunwukonga, 2020-05-27:
To troubleshoot, I would suggest adding the following line at line 1159:
```
print(plot1Cmd,plot2Cmd)
```

Fair warning, as with the other console output, this is two commands printed together. So you'll have to be aware of that when you read it.

leoheck, 2020-05-27:
Sure! 

The lines changed a little bit since my last changes have the original commented

Is it something like this, right?
[image]

leoheck, 2020-05-27:
```
➜ /home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py
Commit1 e80aa7a 	 Update step model for the 0.3.2 version.
Commit2 3454d1a 	 Fix links.
cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show -s --format="%ci" e80aa7 cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show -s --format="%ci" 3454d1
2020-05-25 17:09:56 2020-04-30 15:44:47
Generating .svg files
Traceback (most recent call last):
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py", line 1594, in <module>
    svgDir1, svgDir2, boardDims1, boardDims2 = makeSVG(d1, d2, prjctName, prjctPath)
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py", line 1184, in makeSVG
    return (d1, d2, plotDims1[0], plotDims2[0])
IndexError: list index out of range

```

leoheck, 2020-05-27:
I can't see differences.. can you?

leoheck, 2020-05-27:
Wrong file? I changed it in the kidiff_gui =/

leoheck, 2020-05-27:
Even the lines match now haha.

```
✦ ➜ Commit1 e80aa7a 	 Update step model for the 0.3.2 version.
Commit2 3454d1a 	 Fix links.
cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show -s --format="%ci" e80aa7 cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show -s --format="%ci" 3454d1
2020-05-25 17:09:56 2020-04-30 15:44:47
Generating .svg files
/home/spendless/sandpit/KiCad-Diff/plotPCB.py /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb/plots/e80aa7/yardbird.kicad_pcb /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb/plots/e80aa7 /home/spendless/sandpit/KiCad-Diff/plotPCB.py /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb/plots/3454d1/yardbird.kicad_pcb /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb/plots/3454d1
Traceback (most recent call last):
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py", line 1596, in <module>
    svgDir1, svgDir2, boardDims1, boardDims2 = makeSVG(d1, d2, prjctName, prjctPath)
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py", line 1186, in makeSVG
    return (d1, d2, plotDims1[0], plotDims2[0])
IndexError: list index out of range

[1]  + 256221 exit 1     /home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py
```

sunwukonga, 2020-05-27:
They look OK. The problem must lie in `plotPCB.py`.

leoheck, 2020-05-27:
I don't believe that.
because when I look into them, the content is not the right thing
let me check.

leoheck, 2020-05-27:
Yeah, the files are good. They also work in pcbnew

leoheck, 2020-05-27:
This works too
```
➜ plotPCB.py yardbird.kicad_pcb lala
127199943 62265000 149142914 61785001
```

sunwukonga, 2020-05-27:
But the `svg` files were not generated to the `plots` directory?

leoheck, 2020-05-27:
Not, the command above was made to use the `lala` folder instead

leoheck, 2020-05-27:
The output is good, what should this do?

/home/spendless/sandpit/KiCad-Diff/plotPCB.py
[image]

leoheck, 2020-05-27:
Yeah, it is working, right?

[image]

sunwukonga, 2020-05-27:
That looks good.

leoheck, 2020-05-27:
Only one folder has icons, is it right?

sunwukonga, 2020-05-27:
You only ran one side, not the other side. You'd have to copy paste the other command to get those as well.

leoheck, 2020-05-27:
ah, ok
let me try here

leoheck, 2020-05-27:
Cool, it worked

leoheck, 2020-05-27:
> That's puzzling. I don't see why those commands would work directly, but not when called by Popen.
Indeed!

leoheck, 2020-05-27:
let me learn how to use Popen to test here

sunwukonga, 2020-05-27:
I'm using Python 3.8.2; what version do you have?

leoheck, 2020-05-27:
Just to give more info
➜ python3 --version 
Python 3.8.2

leoheck, 2020-05-27:
ah hehe, same time

leoheck, 2020-05-27:
Do you use stderr? We should print it?

sunwukonga, 2020-05-27:
It wasn't used, but you can.
Add:
```
print("STDERR" + stderr.decode('utf-8'))
```

leoheck, 2020-05-27:
This is the problem

yardbird-0.3.2/pcb on  yardbird-0.3-chamfer [?] 

```
✦ ➜ echo /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb/yardbird.kicad_pcbCommit1 e80aa7a 	 Update step model for the 0.3.2 version.
Commit2 3454d1a 	 Fix links.
cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show -s --format="%ci" e80aa7 cd /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb && /usr/bin/git show -s --format="%ci" 3454d1
2020-05-25 17:09:56 2020-04-30 15:44:47
Generating .svg files
>>>  /home/spendless/sandpit/KiCad-Diff/plotPCB.py /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb/plots/e80aa7/yardbird.kicad_pcb /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb/plots/e80aa7

>>>  /home/spendless/sandpit/KiCad-Diff/plotPCB.py /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb/plots/3454d1/yardbird.kicad_pcb /home/lheck/Dropbox/deepx/YARDBIRD/board/yardbird-0.3.2/pcb/plots/3454d1

out>  b''

err>  b'/bin/sh: 1: /home/spendless/sandpit/KiCad-Diff/plotPCB.py: not found\n'

>  []

out>  b''

err>  b'/bin/sh: 1: /home/spendless/sandpit/KiCad-Diff/plotPCB.py: not found\n'

>  []

Traceback (most recent call last):
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py", line 1613, in <module>
    svgDir1, svgDir2, boardDims1, boardDims2 = makeSVG(d1, d2, prjctName, prjctPath)
  File "/home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py", line 1203, in makeSVG
    return (d1, d2, plotDims1[0], plotDims2[0])
IndexError: list index out of range

[1]  + 265928 exit 1     /home/lheck/Dropbox/kicad-plugins/KiCad-Diff/kidiff_linux.py

```

leoheck, 2020-05-27:
:)
plotProg = '/home/spendless/sandpit/KiCad-Diff/plotPCB.py'

leoheck, 2020-05-27:
And there are the right one! Good Job. Let me try
plotProg = '/usr/local/bin/plotPCB.py'

sunwukonga, 2020-05-27:
Yeah, that's a configurable. Our bad though. It should definitely print the error to show the issue.

leoheck, 2020-05-27:
AWEEEESSOME! hehe

[7 more replies]