Gasman2014/KiCad-Diff #17: Improve Comparison page.

leoheck, 2020-06-01:
Hi, 

Right now, the page with the Layer diff has the `Diff_image`, `Old_version_image` and the `New_version_image` presents the last 2 images as a static image. 

I am thinking that these two images could also be a canvas where it is possible to change the zoom level and board placement. 

Why this? 

I have one layout that has some texts near the board, with info explaining things. This text uses all the space of the board. So being able to zoom in and move the `old_version` and `new_version` images could help review changes even better.

See, how hard it to use these 2 images when the board is big or when it is not possible to move them as it is possible with the main `diff_image` (I can't share this board, this is why I am using the `Eco2.User` to exemplify there, so the text is also is invisible, it is in another layer.)

[image]

Do you think it is a good addition? I believe it is just a matter of reusing the same html that the main `Diff_Image` uses.

Gasman2014, 2020-06-01:
I thought I had left that functionality in - I had it set so that the two bottom windows should pan and zoom together and it was working on testing but I have just checked and the most recent version doesn’t seem to work - sorry.

I agree this was (and should be) the expected functionality. I used the svg-pan-zoom plug in from here http://ariutta.github.io/svg-pan-zoom/demo/sinchronized.html. I did try and get all three windows to pan and zoom together but ran into some limitations of the plugin.

Unfortunately, I do not have any time at present to look at this, I am afraid - I am tied up for the next couple of weeks.

leoheck, 2020-06-01:
> I did try and get all three windows to pan and zoom together but ran into some limitations of the plugin.

Oh, that is a really nice thing.

You don't have to do it or rush to implement this, I was just starting the discussion.

If I need this before you are able to check it I can give a try to implement it too.

Thank you for your reply.

leoheck, 2020-06-01:
Just to add here, I was playing with the demo you added. 

It does not behave that good. Maybe it is my browser (Google Chrome 83.0.4103.6)

For some reason, I can pan upwards/downwards the 2nd tiger but it glitches weird.
Can't even do the same on the 1st tiger. Only the right-left pan is working. 

Maybe this is just a bad config I don't know.

closed 2021-05-06 by leoheck