Gasman2014/KiCad-Diff #22: Crashes when starting webserver

BZab, 2020-12-07:
Hi,

I was trying to run it on Ubuntu (18.04 + python3.6), after generating svgs whole script gets stuck on error:
`serving at port 9092
Gtk-Message: 16:30:43.554: Failed to load module "topmenu-gtk-module"
----------------------------------------
Exception happened during processing of request from ('127.0.0.1', 56782)
Traceback (most recent call last):
  File "/usr/lib/python3.6/socketserver.py", line 320, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 351, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.6/socketserver.py", line 364, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "./KiCad-Diff/kidiff_linux.py", line 1530, in __init__
    super().__init__(*args, directory=os.path.realpath(prjctPath + plotDir), **kwargs)
TypeError: __init__() got an unexpected keyword argument 'directory'
----------------------------------------
`

frankalicious, 2020-12-14:
The directory argument for http.server was introduced with python 3.7.
See https://stackoverflow.com/a/52531444
On my Ubuntu 18.04 I had to change the first line of kidiff_linux.py from  
```
#!/usr/bin/env python3
```
to
```
#!/usr/bin/env python3.7
```

Gasman2014, 2020-12-17:
Does running this under Python>=3.7 solve your problem? If so, I think it would be simplest to make this the minimum version required - I'm not inclined to write ifdefs to accommodate multiple versions.

BZab, 2020-12-30:
Sorry for the delay, I couldn't test it earlier.
In the meantime I have updated python to 3.8.0 and changing sheabang to 3.8 helped.

Gasman2014, 2020-12-31:
Closing. Make minimum python version >=3.7 a requirement to continue to support serving page from arbitrary directory using `SimpleHTTPRequestHandler`.

closed 2020-12-31 by Gasman2014