Gasman2014/KiCad-Diff #31: grep dependency causes silent/inaccurate failure on Windows machines

Cushychicken, 2021-02-09:
I downloaded and ran KiCad-Diff for the first time today, and could not get it to produce a diff of any two PCB files - even ones I knew were radically different. Then I noticed this line:

[image]

Is there some way to remove the `grep` dependency here that will allow this to run smoothly on Windows 10? Or, perhaps, there's another way to run it that won't butt into this issue?

Gasman2014, 2021-02-17:
It may be possible to use FINDSTR in Windows. I will have a look at some point - there may also be a different way to do this altogether. There is a bit of refactoring & development going on at present and this may get shaken out in the process.
I will keep this issue open.

jnavila, 2021-02-17:
I'm working on putting all post processing of the scm commands in Python and remove reliance on external utilities. This way, the system is easier to run across platforms;

jnavila, 2021-02-24:
@Cushychicken could you test with the latest version and report?

Cushychicken, 2021-02-24:
@jnavila sure! May not be until the weekend.

Thanks for looking into this.

leoheck, 2021-03-02:
meanwhile @Cushychicken if the grep command is missing you could use Windows PowerShell or the Linux subsystem for windows.

> I'm working on putting all post-processing of the same commands in Python and remove reliance on external utilities. 
@jnavila this would be nice!

leoheck, 2021-06-18:
@jnavila did you solve this issue? If yes we could close this. What do you think?

jnavila, 2021-06-18:
I did it in #34 and following. This can be closed.

leoheck, 2021-06-18:
Oh, I cannot close it. maybe @Cushychicken or @Gasman2014

Cushychicken, 2021-06-19:
Finally got around to testing - life has been insane these last few months

Fixed - closing

closed 2021-06-19 by Cushychicken

leoheck, 2021-06-19:
@Cushychicken glad to hear that, thanks.