Gasman2014/KiCad-Diff #35: What does this line mean?

jnavila, 2021-02-16:
https://github.com/Gasman2014/KiCad-Diff/blob/335a151e30bc9cc51e6b8d3710887feb5836baf2/kidiff#L122

I don't understand the perl command in this line. If I try it in an svn repo, I only get a presentation string of perl, which is probably not the intended output.

Gasman2014, 2021-02-17:
`cmd = 'cd {prjctPath} && svn log -r HEAD:0 | perl -l40pe "s/^-+/\\n/" | sed "s/ [0-9]\\+ line //g" | sed "s/^ //g" '.format(prjctPath=prjctPath)`
Think you have a typo in the quote but nevertheless its a bit opaque. Pretty sure it was to select the commit message text and trim it to 40 characters (to fit into the Tk box). There is also then some `sed` fu to tidy up some extra junk that came along for the ride. This could probably(?certainly) be simpler and done completely in python +/ RE.
I don't have access to my SVN server at present so I can't easily check this.

I would prefer to do away with the Tk interface completely, TBH and use a web browser interface to live select commits of interest. I will make this a new issue.

jnavila, 2021-02-17:
OK, I'm not sure at all that the perl posprocessing is actually doing that (my citation is directly from the source file, and using it in a term gives me the intended output), but I understand this is in fact a presentation processing which should take place in tkUI.

Thanks for the en

Gasman2014, 2021-02-17:
The reason that I put it in the SVN section rather than the interface is because there is no entirely comparable function in git, fossil and SVN that spits out an identically formatted revision IDs, commit messages etc. The tK layout has limited space.

jnavila, 2021-02-17:
Understood. TBH, I no longer have anything under svn, so getting it correctly is more difficult.

closed 2021-02-19 by jnavila