Gasman2014/KiCad-Diff #38: Small improvement on triptych view

leoheck, 2021-03-02:
Hey @Gasman2014 how easy would be to improve the triptych view by adding 2 buttons to go to the next/previous comparison/layer? This would be a good improvement for the one that is actually reviewing things.

Also, how easy would be to adjust the html to make it autofit the browser. Right now the page is by default bigger than the window.

Gasman2014, 2021-03-04:
I think what is needed first is to retrieve the actual layer names and iterate through these. At present, the stackup is fixed (and even included non-existent layers).

leoheck, 2021-03-04:
Agree, and also, this seems to be easy.

One issue we are going to see doing this is the possibility of changes in the stack up.

For instance, the board starting with 2 layers and ending with 4 layers, and vice versa.
In this case, we need to generate a blank SVG to compare, or just to paint the whole SVG in a different color indicating it was added or removed.

leoheck, 2021-06-18:
I am closing this right now.

closed 2021-06-18 by leoheck