Gasman2014/KiCad-Diff #41: Dates on SVN commits are not being presented.

leoheck, 2021-03-03:
Hey @jnavila I was testing Kidiff with SVN and then I saw that the date-time field is not getting the value from the SVN output.
Do you have any idea on how to improve this?

[image]

jnavila, 2021-03-03:
Can you post the output of:
```bash
svn log --xml -r HEAD:0
``` 
?

leoheck, 2021-03-03:
Sure

First of all, svn version
```
➜  svn --version
svn, version 1.14.0 (r1876290)
   compiled Aug 17 2020, 17:01:19 on x86_64-pc-linux-gnu
```

Here is some extra info. I added 4 tests in case someone wants to use these logs in the future too.

Here a repo with the whole project in the root
```
➜  assoc-board-svn svn log --xml -r HEAD:0

<?xml version="1.0" encoding="UTF-8"?>
<log>
<logentry
   revision="2">
<author>lheck</author>
<date>2021-03-02T23:15:26.786178Z</date>
<msg>Change things</msg>
</logentry>
<logentry
   revision="1">
<author>lheck</author>
<date>2021-03-02T23:14:41.013260Z</date>
<msg>Initial commit</msg>
</logentry>
</log>
```

Here a repo with the whole project in the root and passing the board file to reduce the number of commits
```
➜  assoc-board-svn svn log --xml -r HEAD:0 board.kicad_pcb 

<?xml version="1.0" encoding="UTF-8"?>
<log>
<logentry
   revision="2">
<author>lheck</author>
<date>2021-03-02T23:15:26.786178Z</date>
<msg>Change things</msg>
</logentry>
<logentry
   revision="1">
<author>lheck</author>
<date>2021-03-02T23:14:41.013260Z</date>
<msg>Initial commit</msg>
</logentry>
</log>
```

This project is organized in folders

```
➜  homebase-hardware-svn svn log --xml -r HEAD:0                                    

<?xml version="1.0" encoding="UTF-8"?>
<log>
<logentry
   revision="3">
<author>lheck</author>
<date>2021-03-02T22:34:29.022730Z</date>
<msg>Remove second wire</msg>
</logentry>
<logentry
   revision="2">
<author>lheck</author>
<date>2021-03-02T22:31:35.355701Z</date>
<msg>Update wire</msg>
</logentry>
<logentry
   revision="1">
<author>lheck</author>
<date>2021-03-02T22:30:17.233878Z</date>
<msg>Add initial version of the project</msg>
</logentry>
</log>
```

This project is organized in folders and I am passing an specific board to the SVN command.

```
➜  homebase-hardware-svn svn log --xml -r HEAD:0 led-panel/board.kicad_pcb 

<?xml version="1.0" encoding="UTF-8"?>
<log>
<logentry
   revision="1">
<author>lheck</author>
<date>2021-03-02T22:30:17.233878Z</date>
<msg>Add initial version of the project</msg>
</logentry>
</log>
```

jnavila, 2021-03-03:
Thank you. The mistake was obvious. Will fix shortly.

closed 2021-03-03 by Gasman2014

leoheck, 2021-03-03:
Yay! Thank you
[image]

PR #44 (2021-03-03): Use correct format for date in svn
This should fix #41
commit 69da5d93 (2021-03-03): Use correct format for date in svn

This should fix #41