Gasman2014/KiCad-Diff #49: Y shift between commits

metropt, 2021-03-25:
Hi,
KiCad-Diff is showing a Y shift however if I check x,y positions of the footpirnts between the two commits they are the same.

[image]

The only real difference between this two commits are these pads:
[image]

Any idea why this is happen?

Gasman2014, 2021-03-25:
The current plotting logic depends on the bounding box for the entire design. If items, perhaps on another layer have moved or been added, then the whole plot is shifted. Check your other layers - perhaps something has been altered on the Dwgs or ECO layers?

leoheck, 2021-03-26:
Yeap, true. We could check if it is possible to export the whole page instead. This could fix these minor issues.  And it makes sense since we have a lot of layers to pass through and check for changes.

leoheck, 2021-03-26:
Oh, I found this, and also tested it. Doesn't work yet. Maybe it has to be used with another thing. 

> Do not try to change SetPlotFrameRef(False) to SetPlotFrameRef(true) the result is the pcbnew lib will crash if you try to plot the unknown frame references template.

leoheck, 2021-06-18:
Hello @metropt, could you check if this issue can be closed? If so, feel free to close it, please.

closed 2021-06-18 by metropt