Gasman2014/KiCad-Diff #54: Unable to find board

jorlee99, 2021-05-28:
I am currently running a version of it on a windows distribution it seems that it can no longer find the boards. I have a windows copy of it running just fine on a separate system. This is the error I get.

```
du: cannot access '/home/jordan/TestingKdiff/kidiff/src/6fb3be5/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/TestingKdiff/kidiff/src/6fb3be5/" -q "/home/jordan/TestingKdiff/kidiff/src/6fb3be5/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/PantheonKdiff/KiCad-Diff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/PantheonKdiff/KiCad-Diff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/TestingKdiff/kidiff/src/6fb3be5/expansion.kicad_pcb" for reading
du: cannot access '/home/jordan/TestingKdiff/kidiff/src/d144682/expansion.kicad_pcb': No such file or directory
   plotpcb -o "/home/jordan/TestingKdiff/kidiff/src/d144682/" -q "/home/jordan/TestingKdiff/kidiff/src/d144682/expansion.kicad_pcb"
Traceback (most recent call last):
  File "/home/jordan/PantheonKdiff/KiCad-Diff/plotPCB.py", line 125, in <module>
    processBoard(boardName, plotDir, args.quiet)
  File "/home/jordan/PantheonKdiff/KiCad-Diff/plotPCB.py", line 45, in processBoard
    board = LoadBoard(boardName)
  File "/usr/lib/python3/dist-packages/pcbnew.py", line 6860, in LoadBoard
    return _pcbnew.LoadBoard(*args)
OSError: Unable to open filename "/home/jordan/TestingKdiff/kidiff/src/d144682/expansion.kicad_pcb" for reading
```

Gasman2014, 2021-05-28:
What version of KiCad are you running?

jorlee99, 2021-05-28:
> What version of KiCad are you running?
> […](#)

Kicad 5.1.10

Gasman2014, 2021-05-29:
> … <x-msg://7/#>

I’ll try and look at this next week. I’ve just tried a couple of quick things but I’m getting a segmentation error from the pcbnew PLOT_CONTROLLER which wasn’t happening previously. I recently upgraded to 5.1.10 too but haven’t had need to use KiDiff recently.

As a matter of interest, you mentioned that this is working on another machine. What version of KiCad is running there? If it is working on 5.1.9, this could be a regression. I’m away with limited connectivity.

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   _pcbnew.so                    	0x0000000107073e76 WS_DRAW_ITEM_LIST::BuildFullText(wxString const&) + 406
1   _pcbnew.so                    	0x0000000106fbbf55 WS_DRAW_ITEM_LIST::BuildWorkSheetGraphicList(PAGE_INFO const&, TITLE_BLOCK const&, KIGFX::COLOR4D, KIGFX::COLOR4D) + 2261
2   _pcbnew.so                    	0x0000000106fddf21 PlotWorkSheet(PLOTTER*, TITLE_BLOCK const&, PAGE_INFO const&, int, int, wxString const&, wxString const&, KIGFX::COLOR4D) + 673
3   _pcbnew.so                    	0x0000000106a42945 StartPlotBoard(BOARD*, PCB_PLOT_PARAMS*, int, wxString const&, wxString const&) + 1109
4   _pcbnew.so                    	0x0000000106a3e5aa PLOT_CONTROLLER::OpenPlotfile(wxString const&, PlotFormat, wxString const&) + 506
5   _pcbnew.so                    	0x0000000106ba64f3 _wrap_PLOT_CONTROLLER_OpenPlotfile(_object*, _object*) + 275

leoheck, 2021-06-18:
@jorlee99 this can be closed, right?

leoheck, 2021-06-29:
Hi guys, I am currently closing this since @jorlee99 was having these issues related to my `kdiff` and not to `Kicad-Diff`. The log output posted in the issue was actually the execution of kdiff. Also, we fixed jorlee99 issues on the kidiff repo. But feel free to open the issue again if I am mistaken.

closed 2021-06-29 by leoheck