Gasman2014/KiCad-Diff #56: Help to maintain the repository.

leoheck, 2021-06-18:
@Gasman2014 what do you think about adding @jnavila and me as maintainers here? With more people, we could work faster to improve small bugs.

Gasman2014, 2021-06-19:
I would be very happy for you to be added as maintainers. I am a UK Consultant Anaesthetist by trade, not a programmer and work is very busy and demanding at present (as is family time). I have not really been able to keep this project refreshed as much as I would like. Added to which, even if I get a few mins to look at things whilst at work, my employer has now blocked access to GitHub and also drops my VPN connection so even looking at issues has become arduous.

leoheck, 2021-06-20:
Pretty nice. Thanks for sharing this info. I had no idea that you work in a totally different area. This is pretty interesting and also motivating.

This is something that you could consider doing to add us as collaborators here.

https://docs.github.com/en/enterprise-server@3.0/organizations/managing-access-to-your-organizations-repositories/adding-outside-collaborators-to-repositories-in-your-organization

Ah, by the way. If you want to overcome the limitation in access that your work imposes, you could use ssh to route web traffic securely without a vpn using socks tunnel from a computer working at your work, for instance. You just need a computer running ssh server at home (for instance), a simple ssh command to start this tunnel in the company's computer and set the OS to use the tunnel to access internet. :)

Gasman2014, 2021-06-28:
I've invited you Leandro. I'll wait to hear from @jnavila.

jnavila, 2021-06-29:
I won't be overly active in the coming months, so as long as @leoheck is as quick as he is now, I don't think adding me will be useful in speeding up the development.

Gasman2014, 2021-06-30:
Thats fine, @jnavila. BW

closed 2021-06-30 by Gasman2014