Gasman2014/KiCad-Diff #67: Old branches

leoheck, 2021-07-05:
This repo has some old branches (SVG, and add-license-1). Do you have any special work happening there? @Gasman2014 do you remember what you were doing on this SVG? What if we close these 2 or merge them? Do you have any idea?

Gasman2014, 2021-07-06:
Pretty certain everything I worked on in the SVG branch has been merged to master. (Previously used a different strategy using Inkscape to compare revisions which was way too slow - developed the SVG branch to work on this but haven’t got round to closing it). The add-licence is redundant. Ill take a peek later but think all can be closed.

Gasman2014, 2021-07-10:
The SVG branch was when I ported my original bash script (which used ImageMagick) over to use the SVG comparison method we used today. Then started rewriting it in python alongside the bash script before merging it all to master. 
Ancient history - Its now gone.

closed 2021-07-10 by Gasman2014