Gasman2014/KiCad-Diff #68: Error in pcbnew when calling `plotpcb` (create wxApp before calling this)

edahlseng, 2021-09-16:
I'm trying to run `plotpcb`, but getting an error in what I think is the `pcbnew` library. I'm seeing the following error message printed out multiple times:

```
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
```

Do you know what might be causing this?

Gasman2014, 2021-09-16:
What version of KiCad are you running? I have not been testing this against
5.99 yet

edahlseng, 2021-09-16:
Ah, yes, this is against 5.99.

Gasman2014, 2021-09-16:
Yeah, the Python API has been in a state of flux in 5.99 - tbh I haven't tried to fix this until it was a bit more stable. The diff only needs access to the plotting routines which should be exposed without creating a wx interface, so (hopefully) should be an easyish fix. I'll try and take a look at this a bit later.

leoheck, 2021-09-17:
@edahlseng check if this helps you https://github.com/Gasman2014/KiCad-Diff/issues/55

edahlseng, 2022-01-06:
I tried this again now that KiCad 6 is out, but it looks like it's still failing. I see what I think may be a new error line before the error messages I mentioned earlier (though very similar):
```
Plot1 error: ./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
```

leoheck, 2022-01-06:
OS?

leoheck, 2022-01-06:
It works for me on Linux. I also have some changes to push I am just not sure if they change much. I am also using another solution to fix the generated svg images since the SVG generated by Kicad is not well-formed.

edahlseng, 2022-01-06:
macOS. I think I saw in another issue comment that @Gasman2014 was also having issues with macOS. Interestingly, I think that earlier 5.99 versions of KiCad from a couple of months ago were generating images properly despite the error messages. With v6, that no longer appears to be the case. If anyone has pointers on where to look for this I’d be happy to poke around.

leoheck, 2022-01-06:
plotPCB.py has the macos path hardcoded. It may be related. 
What is the output running this command? Replace the `[board_name]` by yours, please.
```plotpcb [board_name].kicad_pcb```

edahlseng, 2022-01-06:
I'm installing with [this method](https://github.com/edahlseng/homebrew-core/blob/master/Formula/kicad-diff.rb), which runs plotpcb under a virtual environment created from the KiCad bundled python.

Here's what I get when running that command:
```
› plotpcb [board_name].kicad_pcb 
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this

Board version: 20211014

 # ID Name         Filename
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
 1  0 TOP          [board_name]-00-TOP.svg
 2  1 GND1         [board_name]-01-GND1.svg
 3  2 24V & 3_3V   [board_name]-02-24V & 3_3V.svg
 4  3 -10V & 5V    [board_name]-03--10V & 5V.svg
 5  4 GND2         [board_name]-04-GND2.svg
 6 31 BOTTOM       [board_name]-31-BOTTOM.svg
 7 34 B_Paste      [board_name]-34-B_Paste.svg
 8 35 F_Paste      [board_name]-35-F_Paste.svg
 9 36 B_Silkscreen [board_name]-36-B_Silkscreen.svg
10 37 F_Silkscreen [board_name]-37-F_Silkscreen.svg
11 38 B_Mask       [board_name]-38-B_Mask.svg
12 39 F_Mask       [board_name]-39-F_Mask.svg
13 40 Specs        [board_name]-40-Specs.svg
14 44 Edge_Cuts    [board_name]-44-Edge_Cuts.svg
15 45 Margin       [board_name]-45-Margin.svg
16 46 B_Courtyard  [board_name]-46-B_Courtyard.svg
17 47 F_Courtyard  [board_name]-47-F_Courtyard.svg
```

I do see SVGs being created for all of the layers, but they're all blank SVGs.

leoheck, 2022-01-06:
> I do see SVGs being created for all of the layers, but they're all blank SVGs.
Dude, please. What is the name of your board? This does not look right. Review my message. Also review the command, please. 

> If anyone has pointers on where to look for this I’d be happy to poke around.
I feel it happens on linux too. Svg from Kicad plotter is not well-formed I think.
You could file a bug report. 

Try this branch to see if it improves something for you.
https://github.com/leoheck/KiCad-Diff/tree/improve-svg

To try svg optimization you will have to install svgo and have it available in our PATH
And you will have to call kidiff or plotpcb with `-x`

edahlseng, 2022-01-06:
> Also, try this branch to see if it improves something for you. BUT FIRST, try to fix the command above. Your board is not [board_name] unless you are playing a trick on us.

No tricks, I just redacted the board name from my output. Everything I pasted above is an exact copy-and-paste from what I ran, except I replaced my actual board name with `[board_name]`.

leoheck, 2022-01-07:
So, is it working or not?

edahlseng, 2022-01-11:
> Dude, focus, please, is it working or not? Does it help or not? What is the output of the command with the right board name?

Please re-read my previous two messages. The output I pasted above was run on an existing board, I just redacted what I posted. The tool is not working, as only blank SVGs are being output.

leoheck, 2022-01-11:
I see. I don't have a macOS and this looks like a specific issue with macOS. On Linux it kind of working. It is not perfect I think, but it does not give me those messages. Have you reported this on Kicad's GitLab or posted this on Kicad forum?

edahlseng, 2022-01-11:
I have not yet, but I can! Do you have any pointers into this codebase on key lines that I could pull out into a minimally reproducible example to post?

leoheck, 2022-01-11:
Post your issue in the Forum first to start this discussion. Maybe someone knows how to fix this. Or file a report directly in Gitlab exactly as you are doing here. I don't have this issue. I am not a macOS user. If you post the forum post or GitLab issue here, I bet John may joining to the discussion too.

leoheck, 2022-01-16:
Kicad 6.0.1 was releaser. Could you check if this is still happening there?

leoheck, 2022-01-16:
I also see some messages but the files are being generated, as you can see.
[image]

leoheck, 2022-01-16:
This is Kicad's version I have installed `6.0.1-79c1e3a40b~116~ubuntu21.10.1`

leoheck, 2022-01-16:
@edahlseng this branch was improved a bit. It is working for me on a Macbook Air (M1) as you can see the screenshot here
https://github.com/leoheck/KiCad-Diff/tree/improve-svg

[image]

I don't think it needs the new Kicad 6.0.1 but it is to to upgrade it anyways.

Gasman2014, 2022-02-11:
Just been looking at the codebase and see that JPCharras merged this a couple of weeks ago which might address the svg rendering issues. It seems to set the default svg precision to 4 digits in the mantissa and I think the overflow might be causing problems.

At present, I can only get good rendering in Chrome. Firefox is sort of OK (as you show above) and Safari is still broken. setSvgPrecision doesn't seem to work, whatever I do.

```
Use 4 digits in mantissa as default and when exporting SVG.
Allow 0 as line width: this is the right value to draw filled shapes with no outline thickness.
```

https://gitlab.com/kicad/code/kicad/-/blob/master/pcbnew/pcb_plot_params.cpp

I think there will be a 6.0.2 (and prob a 6.0.3 bug fix) - I hope but am not sure if this will make the cut.

FWIW I was also going to reformat everything so that its PEP8 compliant.

leoheck, 2022-02-13:
The fix or workaround from Kicad team would be better because the fix I created is not good at all, it is a nonsmart fix that may fail sometimes. To properly fix it I would be required to use a SVG parser instead. And also, it is being applied with an external script. It would be better to have this implemented in python directly. 

> Allow 0 as line width: this is the right value to draw filled shapes with no outline thickness.

This comment here may be something interesting to play with. The line width function changed its name, and maybe its functionality from the previous Kicad.

@Gasman2014 can we work to merge my last PR before we do this PEP8 thing? I can also apply the PEP8 if you are ok with it, after. I did not do that before because PEP8 is pretty rigorous, and a bit annoying sometimes, but it may support a config file that we can customize its behavior a bit.

leoheck, 2022-02-14:
@Gasman2014 Kicad 6.0.2 is already available https://www.kicad.org/blog/2022/02/KiCad-6.0.2-Release/. Tried here removing my workarounds and it still does not work for Firefox. It may not work for Safari too.

Gasman2014, 2022-02-14:
I have just downloaded it but not having any more success with this yet.

leoheck, 2022-02-14:
Could you update me, when possible, with the issues you are having?

leoheck, 2022-02-14:
Ok, I am going to create a new user on my macOS to see if I can reproduce this.

Now, you are running it a bit differently than what I am expecting.
I recommend you to setup the environment first, and then call kidiff with the `kidiff` (bash script) that is going to be available in your PATH. 

Why? Because it is going to call it using the Python interpreter that comes with Kicad (as we used to do). Maybe that python has something different than the system's interpreter that is indicated by the the `kidiff.py` file

Gasman2014, 2022-02-14:
Yes, I set up a completely new Conda environment.

conda create --name "KD" python=3.7 wxpython

I couldn’t create a Conda environment including wxPython with python >3.7 though. Not sure if that’s significant.
Anyhow, when I activate the KD environment, pythonw works fine.

> Why? Because it is going to call it using the Python interpreter that comes with Kicad (as we used to do). Maybe that python has something different than the system's interpreter that is indicated by the the kidiff.py file

Tried that too - and not working either...

/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3 kidiff.py
This program needs access to the screen. Please run with a
Framework build of python, and only when you are logged in
on the main display of your Mac.

What I thin is worth looking at is how QUICK1’s InteractiveHTMLBOM works.

leoheck, 2022-02-14:
The only thing I could find is that our first line uses env. QUICK1’s uses python3 directly.
Can you test changing this while I don't have the environment ready?
[image]

leoheck, 2022-02-14:
On macos even the Finder cannot preview the right content of the generated file. This is something Kicad team does not know.
[image]

This should be the content of the selected file
[image]

leoheck, 2022-02-15:
@Gasman2014 I improved the file picker a bit that was not pretty good for me on macos. Also, fixed the workaround that fixed svg for macos. It is working better on Safari. But the main diff window is still not being displayed. Still could not close that small window. And I could not recreate your issue here yet.

Gasman2014, 2022-02-15:
Tried that too - no dice. I’ve just reinstalled miniconda and am trying to set up a completely new environment on my new MBP M1 and now have even more issues as conda won’t install wxPython at all.
Might go back to brew.

leoheck, 2022-02-20:
I created a new user. Had to install brew again. I cloned the repo. Loaded the environment with `source env.sh` and could not see the issue. 

This is the best I could do on this mac, since I cannot format it. I have others Macbooks at my work, that I can format but this will take some times, since I am in quarantine recovering from Covid.

But we could try to compare some environment settings. 
Can you share the output of some commands so we can try to identify differences?

This is the output of the command `env`
```bash
TERM_PROGRAM=Apple_Terminal
SHELL=/bin/zsh
TERM=xterm-256color
TMPDIR=/var/folders/58/qpbzfk3n3mn2srm0rrnl1zvm0000gq/T/
TERM_PROGRAM_VERSION=443
TERM_SESSION_ID=98B9926E-4331-403C-B5E0-745BF564A232
USER=lheckk
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.sSXN3Mhhy2/Listeners
PATH=/Users/lheckk/Documents/KiCad-Diff/bin:/Users/lheckk/Documents/KiCad-Diff/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin
__CFBundleIdentifier=com.apple.Terminal
PWD=/Users/lheckk/Documents/KiCad-Diff
XPC_FLAGS=0x0
XPC_SERVICE_NAME=0
SHLVL=1
HOME=/Users/lheckk
LOGNAME=lheckk
OLDPWD=/Users/lheckk/Documents/KiCad-Diff
HOMEBREW_PREFIX=/opt/homebrew
HOMEBREW_CELLAR=/opt/homebrew/Cellar
HOMEBREW_REPOSITORY=/opt/homebrew
MANPATH=/opt/homebrew/share/man::
INFOPATH=/opt/homebrew/share/info:
TK_SILENCE_DEPRECATION=1
LC_CTYPE=UTF-8
_=/usr/bin/env
```

This is `which` python, python3 and pythonw
```bash
lheckk@pipinho KiCad-Diff % which python
/usr/bin/python
lheckk@pipinho KiCad-Diff % which python3
/opt/homebrew/bin/python3
lheckk@pipinho KiCad-Diff % which pythonw 
/usr/bin/pythonw
```

These are the versions of each of these things
```bash
lheckk@pipinho KiCad-Diff % python --version
Python 2.7.18
lheckk@pipinho KiCad-Diff % python3 --version
Python 3.9.10
lheckk@pipinho KiCad-Diff % pythonw --version
Python 2.7.18
```

This compare the paths to see if they are symbolic links and if they point to the same file
```bash
lheckk@pipinho KiCad-Diff % ls -ltr $(which python)
lrwxr-xr-x  1 root  wheel  75 Feb  6 18:22 /usr/bin/python -> ../../System/Library/Frameworks/Python.framework/Versions/2.7/bin/python2.7
lheckk@pipinho KiCad-Diff % ls -ltr $(which python3)
lrwxr-xr-x  1 lheckk  admin  39 Jan 19 03:40 /opt/homebrew/bin/python3 -> ../Cellar/python@3.9/3.9.10/bin/python3
lheckk@pipinho KiCad-Diff % ls -ltr $(which pythonw)
lrwxr-xr-x  1 root  wheel  76 Feb  6 18:22 /usr/bin/pythonw -> ../../System/Library/Frameworks/Python.framework/Versions/2.7/bin/pythonw2.7
```

Launching `kidiff` only, launches the Finder showing only project files. This is nice.
[image]

Then the commit selection window (this is still ugly, but did not try to improve it for Mac, yet)
[image]

Then this is Safari working just fine with the fixed version of the workaround
[image]

Here the triptych
[image]

Gasman2014, 2022-02-20:
> lrwxr-xr-x  1 lheckk  admin  39 Jan 19 03:40 /opt/homebrew/bin/python3 -> ..***@***.***/3.9.10/bin/python3

Thanks Leandro - I’ll look into this later today. Which version of macOS are you running? I’m now on Monterey on an M1 but I should be able to access a MBP running Catalina to compare.

leoheck, 2022-02-20:
Hey John, Here with me it is the same version. M1 running the latest Monterey. I also installed it when the machine was pretty clean. This means I installed `brew` and `wxwidgest` at the same time.
The interesting thing here is that my python3 is the one from homebrew while `python (2.17)` and `pythonw` are pointing to the same binary, so they are the same thing. But KicadDiff uses `python3`. When you have a chance, check from where your `python3` comes.

Gasman2014, 2022-02-20:
Not in front of my machine but I did a time machine transfer from Catalina on MBP to M1. I then reinstalled brew - as the old path was /usr/local and is now /opt/homebrew. I also updated miniconda which is now ~/miniconda and not ~/opt/miniconda. Python3 points to the miniconda version. I have a brew installed Python  which I have tried by simply deactivating conda.

Will check everything again later.

John

leoheck, 2022-02-20:
ah, this seems to be a good test then, forcing use brew's version. let's
see if this helps.

Gasman2014, 2022-02-21:
> exit()

So, with the ‘base’ environment active via conda,  I get the same conda installed Python 3.8.11 by invoking ‘python’, ‘python3' or ‘pythonw'

 base  21:42:14  johnpateman  Applications  KiCad 
 %  which python
/Users/johnpateman/miniconda3/bin/python

 base  21:42:25  johnpateman  Applications  KiCad 
 %  python -V
Python 3.8.11

 base  21:42:30  johnpateman  Applications  KiCad 
 %  which python3
/Users/johnpateman/miniconda3/bin/python3

 base  21:42:35  johnpateman  Applications  KiCad 
 %  python3 -V
Python 3.8.11

 base  21:42:45  johnpateman  Applications  KiCad 
 %  which pythonw
/Users/johnpateman/miniconda3/bin/pythonw

 base  21:42:54  johnpateman  Applications  KiCad 
 %  pythonw -V
Python 3.8.11

If I deactivate conda, I get the system installed python & pythonw version 2.7. I need to specify python3 to get the ‘brew’ installed version.

 base  21:43:03  johnpateman  Applications  KiCad 
 %  conda deactivate

 21:43:17  johnpateman  Applications  KiCad 
 %  which python
/usr/bin/python

 21:43:24  johnpateman  Applications  KiCad 
 %  which python3
/opt/homebrew/bin/python3

 21:43:29  johnpateman  Applications  KiCad 
 %  python -V
Python 2.7.18

 21:43:43  johnpateman  Applications  KiCad 
 %  python3 -V
Python 3.9.10

 21:43:47  johnpateman  Applications  KiCad 
 %  pythonw

WARNING: Python 2.7 is not recommended.
This version is included in macOS for compatibility with legacy software.
Future versions of macOS will not include Python 2.7.
Instead, it is recommended that you transition to using 'python3' from within Terminal.

Python 2.7.18 (default, Jan  4 2022, 17:47:56)
[GCC Apple LLVM 13.0.0 (clang-1300.0.29.10) [+internal-os, ptrauth-isa=deployme on darwin
Type "help", "copyright", "credits" or "license" for more information.

>>> exit
Use exit() or Ctrl-D (i.e. EOF) to exit

leoheck, 2022-02-21:
What does it show for `which pythonw` after disabling conda? Also, I forget to check, which versoion is used when I call kidiff directly. 

Can you add this in your `kidiff.py` and launch it to see what it is going to show.
```
import sys
print(sys.version)
```

I think a good place to put this is right under the  `__main__()`

Gasman2014, 2022-02-21:
%  which pythonw
/usr/bin/pythonw
 22:23:55  johnpateman  Applications  KiCad 
 %  pythonw -V
Python 2.7.18

But we call 

/usr/bin/env python3

I’m going to PM you my env via the Forum as I don’t want to post it here.

Gasman2014, 2022-02-21:
./kidiff.py
3.9.10 (main, Jan 15 2022, 11:40:53)
[Clang 13.0.0 (clang-1300.0.29.3)]
This program needs access to the screen. Please run with a
Framework build of python, and only when you are logged in
on the main display of your Mac.

leoheck, 2022-02-22:
> I’m going to PM you my env via the Forum as I don’t want to post it here.

Fair, you can also remove some lines that are sensitive if you feel that they don't contribute to this issue.

leoheck, 2022-02-22:
Check this line here 
[image]

From here, https://wiki.wxpython.org/wxPythonVirtualenvOnMac

You may have been running an initial build of a tool. You mention that you have synced or upgraded your mac with an old install right? I always do a clean install, and this may be happening because of that. Now, how do we fix this?

leoheck, 2022-02-22:
I am also using the same version
````
➜ kidiff board.kicad_pcb -r
/opt/homebrew/opt/python@3.9/bin/python3.9
3.9.10 (main, Jan 15 2022, 11:40:53) 
[Clang 13.0.0 (clang-1300.0.29.3)]
```

I also added this line `print(sys.executable)` to check from where the interpreter is comming.

leoheck, 2022-02-22:
These are the relevant brew things I have installed to make this work.
```
➜ history | grep brew
    3  brew update
    4  brew upgrade
  204  brew install scour
  325  brew install wxwidgets
  327  brew install wx
  338  brew install wxwidgets
  341  brew install wxpython
  427  brew install librsvg
  432  brew install gsed
  458  brew install dos2unix
```

leoheck, 2022-02-22:
These are all brew things I have installed
```
➜ brew leaves | xargs -n1 brew desc

dockutil: Tool for managing dock items
dos2unix: Convert text between DOS, UNIX, and Mac formats
gh: GitHub command-line tool
git: Distributed revision control system
gnu-sed: GNU implementation of the famous stream editor
librsvg: Library to render SVG files using Cairo
scour: SVG file scrubber
wxpython: Python bindings for wxWidgets
wxwidgets: Cross-platform C++ GUI toolkit
```

leoheck, 2022-02-22:
I would ask you to reinstall python3 too
```
brew uninstall --ignore-dependencies python3
brew reinstall python3 
```

For some reason I got this msg, maybe I have installed it with sudo by mistake. 
I had to fix the issue with the following instructions

```
Error: The following directories are not writable by your user:
/opt/homebrew
/opt/homebrew/share/zsh
/opt/homebrew/share/zsh/site-functions
/opt/homebrew/var/homebrew/locks

You should change the ownership of these directories to your user.
  sudo chown -R $(whoami) /opt/homebrew /opt/homebrew/share/zsh /opt/homebrew/share/zsh/site-functions /opt/homebrew/var/homebrew/locks

And make sure that your user has write permission.
  chmod u+w /opt/homebrew /opt/homebrew/share/zsh /opt/homebrew/share/zsh/site-functions /opt/homebrew/var/homebrew/locks
```

leoheck, 2022-02-22:
Ah, reinstall these 2, too
```
brew resintall wxwidgets
brew reinstall wxpython
```

leoheck, 2022-02-22:
Do not forget these 2
```
brew update
brew upgrade
```

Gasman2014, 2022-02-23:
Yes, have done all of that and had a thorough spring clean of my brew and conda setups. Will investigate further as I have got this at least partially running on Catalina so need to check the exact setup and see if there is something different on M1 Monterey.

The wx panel needs a bit of work and I am hitting a problem again with the paths. If you run KiDiff from outside the checkout of interest, it mucks up the path. I have been running it directly from the KiDiff repo and its is failing after selecting the revisions from the board repository..

leoheck, 2022-02-23:
I don't have any issue related to path on Linux.

This is running **Inside the repo**
```
➜  pwd
/home/lheck/Documents/assoc-board
➜  assoc-board git:(main) ✗ kidiff board.kicad_pcb    

      SCM Selected: git 
  Kicad Board Path: /home/lheck/Documents/assoc-board/board.kicad_pcb
Kicad Project Path: /home/lheck/Documents/assoc-board
         REPO Path: /home/lheck/Documents/assoc-board
 Kicad Project Dir: .
   Board File Name: board.kicad_pcb
        Output Dir: /home/lheck/Documents/assoc-board/.kidiff

Commit 1 (a): board.kicad_pcb
Commit 2 (b): 5be11dd | 2021-12-29 14:58:15 -0300 | Leandro Heck | Update 3d models dir variable

Starting webserver at http://127.0.0.1:9092/web/index.html
(Hit Ctrl+C to exit)
^C
```

This is running **Outside the Repo**
```
➜  assoc-board git:(main) ✗ cd ..
➜  pwd
/home/lheck/Documents/
➜  Documents kidiff assoc-board/board.kicad_pcb    

      SCM Selected: git 
  Kicad Board Path: /home/lheck/Documents/assoc-board/board.kicad_pcb
Kicad Project Path: /home/lheck/Documents/assoc-board
         REPO Path: /home/lheck/Documents/assoc-board
 Kicad Project Dir: .
   Board File Name: board.kicad_pcb
        Output Dir: /home/lheck/Documents/assoc-board/.kidiff

Commit 1 (a): board.kicad_pcb
Commit 2 (b): 5be11dd | 2021-12-29 14:58:15 -0300 | Leandro Heck | Update 3d models dir variable

Starting webserver at http://127.0.0.1:9092/web/index.html
(Hit Ctrl+C to exit)
^C
```

And, this is running **on a completely different path**
```
➜  Documents cd /tmp    
➜  pwd
/tmp
➜  /tmp kidiff ~/Documents/assoc-board/board.kicad_pcb

      SCM Selected: git 
  Kicad Board Path: /home/lheck/Documents/assoc-board/board.kicad_pcb
Kicad Project Path: /home/lheck/Documents/assoc-board
         REPO Path: /home/lheck/Documents/assoc-board
 Kicad Project Dir: .
   Board File Name: board.kicad_pcb
        Output Dir: /home/lheck/Documents/assoc-board/.kidiff

Commit 1 (a): board.kicad_pcb
Commit 2 (b): 5be11dd | 2021-12-29 14:58:15 -0300 | Leandro Heck | Update 3d models dir variable

Starting webserver at http://127.0.0.1:9092/web/index.html
(Hit Ctrl+C to exit)
^C
```

You can see all PATH output from Kica-Diff shows the same address.
Also, you can pass the flag `-r` to make sure the .kidiff folder is being deleted before kidiff starts to make sure the old files are gone.

I have no idea how, but it runs perfectly on my macOS. However, I haven't tested running it from other places like outside of the repo. I still want to get a clean macOS to validate this, or to create a process so you can at least try something different.

Also, I was testing this on Windows this time, and there the GUI looks good as it is on Linux. I have no idea why the GUI is so ugly on macOS too. But if you run with the brew's interpreter it at least inherits the dark theme from the OS.

[image]

closed 2022-09-01 by leoheck