Gasman2014/KiCad-Diff #69: Running on KiCad 5.x on macOS

edahlseng, 2021-09-17:
On macOS, it looks like we have to use KiCad's python executable, which is unfortunately still version 2. Unfortunately, however, it looks like when I try to install the requirements in `requirements.txt` for Python 2, some versions are unavailable (`AwesomeTkinter==2021.1.7`, for example). Given that the Python 3 version with 5.99 [doesn't work at the present moment](#68), are there any other workarounds for macOS? Hopefully I'm missing something silly.

leoheck, 2021-09-17:
The current workaround is to use still use Kicad stable.Why do you need the bleeding edge of Kicad?

edahlseng, 2021-09-17:
@leoheck, KiCad stable includes a bundled Python 2 executable. As far as I can tell, Python 2 is not supported by (some of) the modules at the version numbers listed in the requirements.txt. I don't fully understand why this is needed, but it looks like the [plotpcb](https://github.com/Gasman2014/KiCad-Diff/blob/master/plotpcb) wrapper script specifically uses the KiCad python executable on macOS, which would necessitate the Python 2 modules. If I'm understanding this all correctly, then this means that only macOS will have trouble with KiCad stable at the moment? (Thus my attempt to get KiCad nightly working as a workaround...)

leoheck, 2021-09-19:
Hey @edahlseng thinking here right now. I was able to run this with Kicad 5.99 some months ago (I am almost sure about this). The thing that was not working with 5.99 was `plotgitsh`, actually.

`Requirements.txt` is a hint where you can start looking for used dependencies. but if it does not have the right versions or if the packages have different name, you may have to check the output on command line while executing the tool to see what is missing and the you can try installing something to solve any possible missing package.

Now, I dont remember if I have tested it on OSX because I don't have one. But I do test things on OSX sometimes when I want to make sure a feature is working for everybody.

Gasman2014, 2021-09-19:
Like @leoheck wrote, I'm pretty sure this was working on 5,99 a while back. As of a few weeks ago, KiCad 5.99 has bundled Python 3.8 on macOS. I thought this should have been an easy fix as all that would need doing is passing the new python site-packages directory but I am hitting various problems with invoking pcbnew from an external script. Will continue to have a look.

edahlseng, 2022-01-11:
Given that KiCad 6 is out now, I'm closing this!

closed 2022-01-11 by edahlseng