Gasman2014/KiCad-Diff #75: Kicad v6 (>5.99) needs some love

leoheck, 2021-12-22:
Ok, maybe this is time to see how to improve this tool to actually work better on Kicad 6. It looks everything is the same in the python API as I could understand. However, the generated svg files are not quite equal to the ones generated using Kicad v5. They look off-center and also a weird dot/circle on the top-left of the canvas appears. So, something is still missing in the plot config.  I would accept any idea to fix this.

Gasman2014, 2021-12-22:
I left off working on this for a while as the KiCad api seemed to be constantly changing. There was still a hope of eeschema being scriptable which would have made the schematic side easier. 
Now I’m having all sorts of hassle trying to get any external KiCad scripts running on macOS with problems importing pcbnew even when I explicitly run the KiCad bundled Python and import the site/packages. I can import pcbnew from the KiCad console ok.

You need to think about how you fix a frame of reference between revisions as otherwise a small detail eg adding a dimension will completely realign the layers producing a large visual diff. You probably need to set a special alignment point as you can’t simply zoom to the bounding rectangle. This still won’t help if the user movers the whole board around in the page and I see no good way of dealing with this. I did look at OCV to see if you could use computer vision algorithms to maximally match two layers.

John Pateman

Gasman2014, 2021-12-22:
And one other thing, it would be super helpful if the exported svg included the net name. This would make per net diffs possible. Probably should be a feature request. 

John Pateman

leoheck, 2021-12-22:
> There was still a hope of eeschema being scriptable

How do you know this John? I feel exactly the opposite. It looks, by vague answers on Kicad's forum, that they won't do anything related. Documentation is pretty rare, even for the existing pcbnew api. 

> You need to think about how you fix a frame of reference between...

I agree. I was thinking that exporting the whole layout sheet, including the page border would do the trick. I just couldn't do it by tweaking a bit the current plotpcb.py since it looks there is still something off in this process. 

> it won't help it the user moves the layout ...

yeah, but it is what it is. if the user is working with git he already knows that it is better to not move the layout, especially if he is changing other things too. and if this happens, then it is better to move it first, then commit it, and then do other changes splitting these two moments making things easier to review. Now, for existing projects, a slightly misalignment would paint the whole board so, I don't think that, even if we try to realign, it will be good as we would like.

>  And one other thing, it would be super helpful if the exported svg included the net name. This would make per net diffs possible. Probably should be a feature request.

Is this a bug or an issue introduced during lasts changes on Kicad-diff or is this a totally new feature? If it is the former one, I can try to fix it since I just have to make some tests. Now, if this is a new feature and depends on Kicad output, well then it will require more investigation. I am not good in decifering Kicad's python api yet.

leoheck, 2021-12-23:
This is my first try to fix svg's canvas using inkscape on the command line. The result is awesome. Which is also good to review if page settings have changed. The initial issue is that it takes a lot of time. This is Kicad v6 by the way. And then it may have the same result on v5 too since I can use the same settings on plotpcb.py for both. 

[image]

Now, I think that the best way would put this on plotpcb.py is if I find a good way of doing this with python. It may be faster.

Gasman2014, 2021-12-23:
My very first version of KiDiff used Inkscape (it should still be in the repo). If took forever … 🙁

John Pateman

leoheck, 2022-02-14:
I am closing this since this is a Kicad bug.

closed 2022-02-14 by leoheck