Gasman2014/KiCad-Diff #76: Tkinter GUI is not readable

leoheck, 2022-01-16:
Any idea on how to fix this issue, easily, for everyone?

[image]

leoheck, 2022-01-17:
I saw this. But I could just find this solution
[image]

Which involves in replacing the current python. My python on Monterey is 8.9.6 if I am not mistaken. I am not a macOS user, I believe this is not a good solution. But if you are fine with this solution I will try to apply it to check how it is going to work.

Gasman2014, 2022-01-17:
> Which involves replacing the current python. My python on Monterey is 8.9.6 if I am not mistaken. I am not a macOS user, I believe this is not a good solution. But if you are fine with this solution I will try to apply it to check how it is going to work.

I’m not sure its a good solution either!! In fact, It sounds like a fairly bad solution!

TBH. I wonder if it might be better long-term to use wx? After all, that’s what KiCad uses. I chose TKInter to begin with as it seemed a bit easier to use - for a very minimal interface. I have no experience with wx.

leoheck, 2022-01-17:
Ok, if we find a designer tool, to help create wx interfaces we could copy the current one.
To be honest I also have no experience with GUI either haha

leoheck, 2022-01-19:
Initial test using wxWidgets https://github.com/Gasman2014/KiCad-Diff/pull/78

closed 2022-03-09 by leoheck