Gasman2014/KiCad-Diff #82: NotADirectoryError when using kidiff

ZhuYanzhen1, 2022-02-09:
After I select the version, error pop out as below.
```
$ kidiff ../MM32_CLion/circuit/project.kicad_pcb 

      SCM Selected: git 
  Kicad Board Path: /home/laozhu/Desktop/MM32_CLion/circuit/project.kicad_pcb
Kicad Project Path: /home/laozhu/Desktop/MM32_CLion/circuit
         REPO Path: /home/laozhu/Desktop/MM32_CLion
 Kicad Project Dir: circuit
   Board File Name: project.kicad_pcb
        Output Dir: /home/laozhu/Desktop/KiCad-Diff/kidiff

Commit 1 (a): b92d08f | 2022-02-09 16:22:42 +0800 | Lao·Zhu | Switched the GPS serial port location over and updated the PCB file.
Commit 2 (b): 19e482f | 2022-01-18 02:29:55 +0800 | Lao·Zhu | Complete the PCB drawing and DRC check.
Traceback (most recent call last):
  File "/home/laozhu/Desktop/KiCad-Diff/kidiff", line 814, in <module>
    commit1, commit2, commit_datetimes = scm.get_boards(kicad_pcb_path, repo_path, kicad_project_dir, board_filename, commit1, commit2)
  File "/home/laozhu/Desktop/KiCad-Diff/scms/git.py", line 44, in get_boards
    os.makedirs(outputDir1)
  File "/usr/lib/python3.8/os.py", line 223, in makedirs
    mkdir(name, mode)
NotADirectoryError: [Errno 20] Not a directory: '/home/laozhu/Desktop/KiCad-Diff/kidiff/b92d08f'
```bash

ZhuYanzhen1, 2022-02-09:
Oh, I solved the problem. Just need to specify the output path with the -o option when using it.

Gasman2014, 2022-02-09:
If running from the command line, try specifying the output directory with an -o flag. Usually, I make the diffs in the project directory.

There was a recent change which might have broken this.

leoheck, 2022-02-10:
The `-o` is not required, it is optional. By default, KiDiff has to generate the `kidiff` folder in the root path of the repository, but it looks like your project is in a nested folder `circuit`. The issue may be related to that. However, I also have a repo like this and did not see this issue on my repo. I don't have my computer with me, but I can check this in a week or so. 

@ZhuYanzhen1 could you share this repo with me, so I can try to reproduce this issue?

ZhuYanzhen1, 2022-02-10:
@leoheck 
Okay, here's the link to my repository. It's a private repository, so I'll probably convert it to private after you clone it locally.
https://github.com/ZhuYanzhen1/MM32_CLion

leoheck, 2022-02-10:
Ok, thanks. I've cloned it. You can revert it to private again.

leoheck, 2022-02-14:
Hey @ZhuYanzhen1 your issue is solved? 

I am running it here, with Kicad `6.0.2-378541a8eb~116~ubuntu21.10.1` and it looks ok. I did not need to use the `-o` flag but I am running it inside the circuit folder `/home/lheck/Documents/MM32_CLion/circuit`

Ah, I am running the version that is in a pull requesto here in this repo. This version here https://github.com/leoheck/KiCad-Diff/tree/wxwidges_and_pcm But it may work the same as the latest master in this repo here.

```
➜  circuit git:(master) kidiff project.kicad_pcb 

      SCM Selected: git 
  Kicad Board Path: /home/lheck/Documents/MM32_CLion/circuit/project.kicad_pcb
Kicad Project Path: /home/lheck/Documents/MM32_CLion/circuit
         REPO Path: /home/lheck/Documents/MM32_CLion
 Kicad Project Dir: circuit
   Board File Name: project.kicad_pcb
        Output Dir: /home/lheck/Documents/MM32_CLion/circuit/.kidiff
```

However, I am seeing a warning that I saw before. I have no idea what it means yet.
```
19:28:19: Warning: Overflow KiROUND converting value 44548886938008567152640000.000000 to i
```

Here some screenshots to tease you.
[image]
[image]

ZhuYanzhen1, 2022-02-14:
&nbsp;&nbsp;&nbsp;&nbsp;@leoheck It is true that you do not need to add the `-o` parameter with opening the terminal in the project's folder. But despite this, I get a blank display after opening the site, so I'm not sure what's going on. Here's my screenshot:

[image]

&nbsp;&nbsp;&nbsp;&nbsp;And I'm using Kicad `6.0.2-378541a8eb~116~ubuntu21.10.1`. using `kidiff project.kicad_pcb` in the terminal also generates many warnings like ` Warning: Overflow KiROUND converting value 44548886938008567152640000.000000 to i`.

leoheck, 2022-02-14:
Which browser are you using?
It should work on Chrome and Firefox. It may not work on Safari or Edge. I could not test this on these, yet.
Also, what is your Operating System?

leoheck, 2022-02-14:
>   Checking the generated kidiff folder, the generated svg image files for each layer exist, but they just don't show up on the web page.

Ah, then this may be the issue we are having with Kicad. It is generating svg files that are not good on some web browsers. I made a workaround that works on Chrome and Firefox. It is not perfect, but at least it should shows something.

ZhuYanzhen1, 2022-02-14:
> Which browser are you using? It should work on Chrome and Firefox. It may not work on Safari or Edge. I could not test this on these, yet. Also, what is your Operating System?

The browser I am using is the firefox browser that comes with Ubuntu, its version is 96.0 (64bit). My operating system is Ubuntu 20.04.3 LTS, and the Linux kernel version is 5.13.0-30-gernic.

leoheck, 2022-02-14:
My Firefox is 97.0 but it may not be the culprit.  

Make sure you are using this version of Kicad here:
From this repo: https://github.com/leoheck/KiCad-Diff.git
This branch: `wxwidges_and_pcm`

ZhuYanzhen1, 2022-02-14:
The version of Firefox browser is really not the cause of the failure to display, I tried using 97.0 and still get this error. After I switched to the `wxwidges_and_pcm branch`, using the kidiff command after passing the source command reports an error. The error is reported as follows:
```bash
Command kidiff not found
```

ZhuYanzhen1, 2022-02-14:
```
Cookie “pan_zoom_diff_panx” will be soon rejected because it has the “SameSite” attribute set to “None” or an invalid value, without the “secure” attribute. To know more about the “SameSite“ attribute, read https://developer.mozilla.org/docs/Web/HTTP/Headers/Set-Cookie/SameSite
```

leoheck, 2022-02-14:
> After I switched to the wxwidges_and_pcm branch, using the kidiff command after passing the source command reports an error. The error is reported as follows:

My branch changed repo organization a bit.
Please, update your environment with this setup:

```
# Kiri environment setup
# eval $(opam env)
# export TK_SILENCE_DEPRECATION=1
KIRI_HOME=/home/lheck/Documents/kiri
export PATH=${KIRI_HOME}/submodules/KiCad-Diff/bin:${PATH}
export PATH=${KIRI_HOME}/bin:${PATH}
```

Adjust the `KIRI_HOME` accordingly with your installation path

leoheck, 2022-02-14:
Use this command to plot layers of the current board. Unfortunately, I don't have your project anymore. I deleted it after I could check if kidiff was working.

```
plot_kicad_pcb board.kicad_pcb 
```

Check the command line if it shows any issue.
It should look something like this
```
➜  assoc-board git:(main) ✗ plot_kicad_pcb board.kicad_pcb -f

Board 20211014
Kicad v6

 # ID Name          Filename
 1  0 F_Cu          board-00-F_Cu.svg
 2  1 Gnd           board-01-Gnd.svg
 3  2 Pwr           board-02-Pwr.svg
 4 31 B_Cu          board-31-B_Cu.svg
 5 32 B_Adhesive    board-32-B_Adhesive.svg
 6 33 F_Adhesive    board-33-F_Adhesive.svg
 7 34 B_Paste       board-34-B_Paste.svg
 8 35 F_Paste       board-35-F_Paste.svg
 9 36 B_Silkscreen  board-36-B_Silkscreen.svg
10 37 F_Silkscreen  board-37-F_Silkscreen.svg
11 38 B_Mask        board-38-B_Mask.svg
12 39 F_Mask        board-39-F_Mask.svg
13 40 User_Drawings board-40-User_Drawings.svg
14 41 User_Comments board-41-User_Comments.svg
15 42 User_Eco1     board-42-User_Eco1.svg
16 43 User_Eco2     board-43-User_Eco2.svg
17 44 Edge_Cuts     board-44-Edge_Cuts.svg
18 45 Margin        board-45-Margin.svg
19 46 B_Courtyard   board-46-B_Courtyard.svg
20 47 F_Courtyard   board-47-F_Courtyard.svg
21 48 B_Fab         board-48-B_Fab.svg
22 49 F_Fab         board-49-F_Fab.svg
```

And then check if the generated images are good.
If they are good, try to open them directly in your browser to check if they can be displayed.

leoheck, 2022-02-14:
Now, talking about the weird warning about the Kibound. I could find another report with the same sentence. I am adding it here for reference to save it for later.

https://gitlab.com/kicad/code/kicad/-/merge_requests/325#note_391041093

ZhuYanzhen1, 2022-02-14:
> My branch changed repo organization a bit. Please, update your environment with this setup:
> ```
> # Kiri environment setup
> # eval $(opam env)
> # export TK_SILENCE_DEPRECATION=1 [...]

@leoheck It works fine now!

leoheck, 2022-02-14:
Awesome.

closed 2022-02-14 by leoheck