Gasman2014/KiCad-Diff #83: Idea - Button to disable the filled copper areas.

leoheck, 2022-02-26:
It would be good to have a button that hides or removes the fill of the copper areas. This would help to see changes better.

ssarna1, 2022-08-16:
As far as i know, gerbers don't have information which part of copper is area and which is just a trace.

leoheck, 2022-08-16:
Yeah. however, Kicad diff doesn't use Gerber files. It uses .svg files exported/plotted with Kicad. I was thinking that those .svg files could have some kind of info we could use. For instance, maybe polygons of copper fill are drawn as polygons and traces as lines.

Gasman2014, 2022-08-17:
You can examine the svg with a text editor and see that there are fills with zero stroke widths and others with well defined widths,  fills & opacities.

<snip>
<g style="fill:#000000; fill-opacity:1.0; 
stroke:#000000; stroke-width:0; stroke-opacity:1; 
stroke-linecap:round; stroke-linejoin:round;">
<circle cx="70033.6" cy="38467.2" r="196.85" /> 
<circle cx="68458.8" cy="37837.2" r="196.85" /> 
<circle cx="66726.5" cy="38467.2" r="196.85" /> 
<circle cx="66096.6" cy="40120.7" r="196.85" /> 
<circle cx="70033.6" cy="41774.2" r="196.85" /> 
</g>
<g style="fill:#000000; fill-opacity:1.0; 
stroke:#000000; stroke-width:393.701; stroke-opacity:1; 
stroke-linecap:round; stroke-linejoin:round;">
</g>
<g style="fill:#000000; fill-opacity:0.0; 
stroke:#000000; stroke-width:393.701; stroke-opacity:1; 
stroke-linecap:round; stroke-linejoin:round;">
<path d="M39718 33199
L41852 33199
</snip>

I am not sure how fruitful it would be to try to identify features this way other than helping to identify the co-ordinates of a feature diff. What would be more useful would be to identify contiguous areas that correspond to nets of connectivity so that you can identify nets where features have changed.

There is an outstanding GitLab issue & feature request to tag elements within the .svg with the net name. This would be helpful and I suspect not a major or breaking change.

https://gitlab.com/kicad/code/kicad/-/issues/2196

ssarna1, 2022-08-18:
> Yeah. however, Kicad diff doesn't use Gerber files. It uses .svg files exported/plotted with Kicad. I was thinking that those .svg files could have some kind of info we could use. For instance, maybe polygons of copper fill are drawn as polygons and traces as lines.

You're right, i had a brainfart from looking at .svg's for too long.

leoheck, 2022-08-18:
I see. So how can we help you?

closed 2022-09-11 by leoheck