Gasman2014/KiCad-Diff #84: wxWidgets issues on macOS

leoheck, 2022-03-09:
Hey @Gasman2014 I could find a Mac with issues on wx. I don't know it the problem is pretty the same as you are having. But I could fix it there with this.

I removed and reinstalled homebrew just to make sure it was not pretty old or bad somehow.
```
# List of installed brew stuff to install them back later
brew leaves | xargs -n1 brew desc

# Remove old homebrew installation
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/uninstall.sh)"

# Manually remove folders that were not removed (maybe due to some old/wrong permissions)
sudo rm -rf /usr/local/Caskroom
sudo rm -rf /usr/local/Cellar
```

Then I have installed back my previously listed brew modules
```
# Install Homebrew again
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"

# Install old brew modules
brew install dockutil
brew install dos2unix
brew install gh
brew install git
brew install gnu-sed
brew install librsvg
brew install scour

# These two guys are enough for putting wx to work
brew install wxwidgets
brew install wxpython
```

Then I have added this on top of the `kidiff.py` but you can test my PR (here https://github.com/Gasman2014/KiCad-Diff/pull/81) directly since it is carrying this change already.
```
# Test, needed for wx sometimes for macos (this is just a test, a better solution is comming it this works)
if platform.system() == "Darwin":
    sys.path.insert(0, "/usr/local/lib/python3.9/site-packages/")
```

Then Kicad-Diff with `wx` worked again on that Mac (which is a Mac Mini with Monterey that was a dirty upgrade).

leoheck, 2022-03-22:
Hi @Gasman2014 could you check if this fix works for you?

Gasman2014, 2022-03-22:
I have a wx demo working now on this machine so wx seems to work OK but here I am getting this.

Traceback (most recent call last):
  File "/Users/johnpateman/Documents/Code/KiCad-Diff/kidiff/./kidiff.py", line 17, in <module>
    from kidiff_gui import commits_dialog
  File "/Users/johnpateman/Documents/Code/KiCad-Diff/kidiff/kidiff_gui.py", line 4, in <module>
    from kidiff_wxglade import CommitsDialog
  File "/Users/johnpateman/Documents/Code/KiCad-Diff/kidiff/kidiff_wxglade.py", line 15, in <module>
    class CommitsDialog(wx.dialog):
AttributeError: module 'wx' has no attribute ‘dialog'

leoheck, 2022-03-22:
How are you launching the app?
Are you using the `kidifff` that is inside the bin folder (`./KiCad-Diff/bin/kidiff`)?

leoheck, 2022-03-23:
This is crazy. I could install this wx on every mac I have access.

Ah, there is also a wx window test, in this `./KiCad-Diff/kidiff/` folder.
This script `./kidiff_gui.py`. Could you check if it gives you the same result?

Gasman2014, 2022-03-23:
Yup, crazy. I’m very tempted to wipe and start again but I have a lot of stuff thats tricky and time consuming to recompile. I now have a older MBP that might be a good option to start from scratch on. 

John Pateman

Gasman2014, 2022-03-23:
> I understand, totally. Have you tried to create a new user to test this? It
> won't be a fresh start but it be something different.

I figured that it wasn’t primarily the stuff in userspace would be where the problem was...

leoheck, 2022-03-24:
John!

I found this file picker using wx.
I want to replace all the file pickers I am using buy something like this.
You can also use this code to check if it work on your mac, I just run it on linux and it is a good replacement for the other things.

```python
#!/usr/bin/python

import wx

def onButton(event):
    print "Button pressed."

app = wx.App()

frame = wx.Frame(None, -1, 'win.py')
frame.SetDimensions(0,0,200,50)

# Create open file dialog
openFileDialog = wx.FileDialog(frame, "Open", "", "", 
      "Python files (*.py)|*.py", 
       wx.FD_OPEN | wx.FD_FILE_MUST_EXIST)

openFileDialog.ShowModal()
print(openFileDialog.GetPath())
openFileDialog.Destroy()
```

leoheck, 2022-03-24:
Actually, this would be enough

```python
#!/usr/bin/python

import pathlib
import wx

def open_project_dialog(current_path):

    title = "Select Kicad Project"

    file_wildcards = "Kicad Project (*.pro;*.kicad_pro)|*.pro;*.kicad_pro|All files (*)|*"

    openFileDialog = wx.FileDialog(
        frame, title, "", "", file_wildcards,
        wx.FD_OPEN | wx.FD_FILE_MUST_EXIST)

    openFileDialog.ShowModal()

    kicad_project_path = openFileDialog.GetPath()
    openFileDialog.Destroy()

    return kicad_project_path

if __name__ == "__main__":

    app = wx.App()
    frame = wx.Frame(None, -1, 'win.py')
    frame.SetDimensions(0, 0, 200, 50)

    current_path = pathlib.Path().absolute()

    kicad_project_path = open_project_dialog(current_path)
    print(kicad_project_path)

```

Gasman2014, 2022-03-25:
Working - but only if I set the shebang to 

#!/usr/bin/env pythonw

Will investigate further with the kidiff script - and dig out the old laptop to compare.
BW

leoheck, 2022-03-25:
This is the first positive result! Nice.

closed 2022-09-01 by leoheck