Gasman2014/KiCad-Diff #85: kidiff not working with SVN

ssarna1, 2022-08-16:
I was having an issue in which the script was not able to check out any revision from SVN (it was reading local copy just fine)
this is the output i had:
```
Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2
 Cmd: svn cat -r r210 /
svn: E155007: '/' is not a working copy

Code: 1

Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2
 Cmd: svn log -r r210

Code: 0

Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2/.kidiff/local
 Cmd: plot_kicad_pcb schemat.kicad_pcb

Code: 0

Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2/.kidiff/r210
 Cmd: plot_kicad_pcb schemat.kicad_pcb

Code: 1

Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2/.kidiff
 Cmd: diff r210/schemat.kicad_pcb local/schemat.kicad_pcb

Code: 1

Starting webserver at http://127.0.0.1:9092/web/index.html
(Hit Ctrl+C to exit)
```

so i went into /kidiff/scms/svn.py and changed those lines:
```
 if not commit1 == board_filename:
            #svnArtifact1 = ["svn", "cat", "-r", artifact1, svnPath]  #removed
            svnArtifact1 = ["svn", "cat", "-r", artifact1, board_filename]  #added

        if not commit2 == board_filename:
            #svnArtifact2 = ["svn", "cat", "-r", artifact2, svnPath]  #removed
            svnArtifact2 = ["svn", "cat", "-r", artifact2, board_filename]  #added
```

now it still gives diff error:
```
Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2
 Cmd: svn cat -r r204 schemat.kicad_pcb

Code: 0

Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2
 Cmd: svn log -r r204

Code: 0

Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2/.kidiff/local
 Cmd: plot_kicad_pcb schemat.kicad_pcb

Code: 0

Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2/.kidiff/r204
 Cmd: plot_kicad_pcb schemat.kicad_pcb

Code: 0

Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2/.kidiff
 Cmd: diff r204/schemat.kicad_pcb local/schemat.kicad_pcb

Code: 1

Starting webserver at http://127.0.0.1:9092/web/index.html
(Hit Ctrl+C to exit)

```

but at least web interface shows both boards and XOR between them, so i can use the tool as usual.

I don't use git i don't even code in python at all. Ii have no clue how to make pull request or if even this problem concerns only me or if anyone else had this. If so, i hope this helps.

Cheers

leoheck, 2022-08-16:
Hi, which version of Kicad-Diff do you have?
You can check this, going inside the folder of the repo and then using the following command:
```bash
git log  -n 1 | cat
```

ssarna1, 2022-08-18:
It's SVN repo, so Your command is not working. But anyway it's pretty fresh download.

leoheck, 2022-08-18:
When did you download it? 2 days ago? a year?

ssarna1, 2022-08-18:
A year old is not pretty fresh by my book (though nobody have released my book yet, so there's that). It was the same day i've posted the issue.

leoheck, 2022-08-18:
Alright, so this means we are talking about this version `5712aef2702430e6f96a432a481f5116a754168c`
I will give it a look for you later once I set up a working subversion server.

leoheck, 2022-09-01:
Hey @ssarna1 could you test this version here to see if it fixes svn for you?
https://github.com/leoheck/KiCad-Diff/tree/svn_fix

ssarna1, 2022-09-07:
Sure:
````
$ kidiff
Traceback (most recent call last):
  File "/home/ls/Programy/Kidiff_test/bin/../kidiff/kidiff.py", line 27, in <module>
    import scms.svn as svn
  File "/home/ls/Programy/Kidiff_test/kidiff/scms/svn.py", line 85
    cmt = (dateTime.splitlines()[1]).split("|")_, SVNdate1, SVNtime1, SVNutc, *_ = cmt[2].split(" ")
                                               ^
SyntaxError: invalid syntax

leoheck, 2022-09-07:
Ops, my mistake. Could you update your repo again and test it one more time?

ssarna1, 2022-09-10:
```
$ kidiff

(kidiff.py:87344): Gtk-WARNING **: 20:23:39.211: Failed to measure available space: Zły argument

      SCM Selected: svn 
  Kicad Board Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2/schemat.kicad_pcb
Kicad Project Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2
         REPO Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2
 Kicad Project Dir: .
   Board File Name: schemat.kicad_pcb
        Output Dir: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2/.kidiff

Commit 1 (a): schemat.kicad_pcb
Commit 2 (b): r186 | 2022-04-20 20:50 | 

Traceback (most recent call last):
  File "/home/ls/Programy/Kidiff_test/bin/../kidiff/kidiff.py", line 826, in <module>
    commit1, commit2, commit_datetimes = scm.get_boards(kicad_pcb_path, repo_path, kicad_project_dir, board_filename, commit1, commit2)
  File "/home/ls/Programy/Kidiff_test/kidiff/scms/svn.py", line 107, in get_boards
    return artifact1, artifact2, time1 + " " + time2
UnboundLocalError: local variable 'time2' referenced before assignment
```
Zły argument => bad argument

leoheck, 2022-09-10:
I see, we are almost there. Thanks for your feedback. I've pushed a fix. Could you check for me, once more? @ssarna1

ssarna1, 2022-09-11:
It seems to be working fine, but it throws some errors:
```
$ kidiff

(kidiff.py:91168): Gtk-WARNING **: 10:41:31.525: Failed to measure available space: Zły argument

      SCM Selected: svn 
  Kicad Board Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2/schemat.kicad_pcb
Kicad Project Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2
         REPO Path: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2
 Kicad Project Dir: .
   Board File Name: schemat.kicad_pcb
        Output Dir: /home/ls/Dokumenty/SVN_cad/Mieszkanie2/trunk2/.kidiff

Commit 1 (a): schemat.kicad_pcb
Commit 2 (b): r172 | 2022-04-14 19:23 | DO

Starting webserver at http://127.0.0.1:9092/web/index.html
(Hit Ctrl+C to exit)
----------------------------------------
Exception occurred during processing of request from ('127.0.0.1', 52024)
Traceback (most recent call last):
  File "/usr/lib/python3.10/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.10/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.10/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/home/ls/Programy/Kidiff_test/bin/../kidiff/kidiff.py", line 648, in __init__
    super().__init__(
  File "/usr/lib/python3.10/http/server.py", line 651, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3.10/socketserver.py", line 747, in __init__
    self.handle()
  File "/usr/lib/python3.10/http/server.py", line 425, in handle
    self.handle_one_request()
  File "/usr/lib/python3.10/http/server.py", line 413, in handle_one_request
    method()
  File "/usr/lib/python3.10/http/server.py", line 658, in do_GET
    self.copyfile(f, self.wfile)
  File "/usr/lib/python3.10/http/server.py", line 858, in copyfile
    shutil.copyfileobj(source, outputfile)
  File "/usr/lib/python3.10/shutil.py", line 198, in copyfileobj
    fdst_write(buf)
  File "/usr/lib/python3.10/socketserver.py", line 826, in write
    self._sock.sendall(b)
BrokenPipeError: [Errno 32] Broken pipe
----------------------------------------
```
Doesn't show any stats but thickness:

Modules | 0 | 0
-- | -- | --
Drawings | 0 | 0
Version | local | r172 | 
Date | 2022-08-16 | 2022-04-14 | 
Time | 20:01:03 | 21:23:15 | 
Nets | 0 | 0
Tracks | 0 | 0
Zones | 0 | 0

But it's enough for my usage. Some layers have pretty dark colours, but in XOR window it's always blue and red, so good enough.
Thanks for the effort, svn seems to be abandoned a lot nowadays.

By the way, do You know any tool that would do the same for schematics and works on Linux? I could only find some tools for Git or for Windows.

leoheck, 2022-09-11:
Strange, it looks like it is using the wrong port

Here it says por 9092
```
Starting webserver at http://127.0.0.1:9092/web/index.html
(Hit Ctrl+C to exit)
```

Here, it looks like this 52024 is a port and it is wrong.
```Exception occurred during processing of request from ('127.0.0.1', 52024)```

> Doesn't show any stats but thickness:

I didn't understand this comment above

> But it's enough for my usage. Some layers have pretty dark colours, but in XOR window it's always blue and red, so good enough.

I would say that the only XOR view that matters.

> Thanks for the effort, svn seems to be abandoned a lot nowadays.

I don't see anyone using SVN since some 12 years ago. And at that time SVN was a something really bad, it was only better than the shitty CSV or something. 

I recommend you to just convert this SVN repo to git. You can do it everytime you need to use kidiff.

> By the way, do You know any tool that would do the same for schematics and works on Linux? I could only find some tools for Git or for Windows

Sure, the plotkicadsh/plotgitsch is a tool for that. It works for Kicad 5 files. 

I have a tool https://github.com/leoheck/kiri that works on git repos that creates a really good way to review schematic and layout. It uses kidiff and plotgitsch. But it also make it possible to review Kicad 6 files. 

In your case you can use it if you convert your SVN to a hit repo.

leoheck, 2022-09-11:
I am closing this issue since the fix was merged. Thanks @ssarna1 
https://github.com/Gasman2014/KiCad-Diff/pull/87

closed 2022-09-11 by leoheck

ssarna1, 2022-09-11:
> I didn't understand this comment above

I meant stats like: Nets, tracks, zones etc. are shown as 0, if that's the case then it would be better to not display it at all, it just takes screen space with no benefits..

> I recommend you to just convert this SVN repo to git. 

Yeah, everyone is saying that, but SVN was dead simple to set up on raspberry pi, path based authentication is dead simple as well, and i want to be able to check out just part of repo to work on, and as far as i know the last one is not possible with git. So for foreseeable future i'll stick with SVN (VS code has a plugin at least, and it's more needed with code than schematics)

Thanks again!

leoheck, 2022-09-11:
> I meant stats like: Nets, tracks, zones etc. are shown as 0, if that's the case then it would be better to not display it at all, it just takes screen space with no benefits..

This should be fixed, but since I don't use this information, I did not try to do it. Someone interested could help. 

> Yeah, everyone is saying that, but SVN was dead simple to set up on raspberry pi, path based authentication is dead simple as well, and i want to be able to check out just part of repo to work on, [...]

This seems completely wrong. Even if you check the whole repo, you can delete parts of it after doing that. But look for space checkout. 

You can also install `gh cli tool https://lindevs.com/install-github-cli-on-raspberry-pi easily to setup the authentication easily with a single command. 

There is no reason at all to use outdated SVN.

Also, make yourself a favor, use something better than VS. This shit is just slow (I use Sublime Text)

Alright, the PR was merged fixing the SVN issues, thanks for your help. I could setup an SVN repo on my computer a few minutes ago and everything is working as intended (except by the info you already know). 

The port issue seems to be something from your side. I did not have this here as you can see my logs.

[image]