Gasman2014/KiCad-Diff #86: Pleas create addon plugin

RolandWa, 2022-08-26:
Hi All,

Nice to work on the routing tools. Please add this tools to the KiCad plugin manager using description from
https://dev-docs.kicad.org/en/addons/

Best Regards
Roland

Gasman2014, 2023-01-23:
Hi Roland,
I have considered adding this as a KiCad plugin so that it works a bit like the interactiveHTMLBom plugin from quick1. I don't think making it a plugin would be terribly difficult, however, I don't have a lot of time at present and I suspect there would be quite a lot of support requests as the toolchain - esp plotgitsch is not straightforward.

I think it would be more useful to work on converting the plotting to use the new kicad-cli as this would make the install easier and remove the more complex dependencies. This would limit the functionality to KiCad 7 but I don't think that would be a major problem and I really don't have enough time to maintain multiple versions.

leoheck, 2023-01-23:
Hey @Gasman2014 , I made the kiri a plugin, it was not accepted due to many (thousands) dependencies that were not python related. But doing this with kidiff would be possible, and I think they may accept it since it uses python only.

If you wanna try, you can check my plugins here, https://github.com/leoheck/kiri/tree/main/kicad

Gasman2014, 2023-01-23:
Yes, I think the use of kicad-cli should reduce the need for so many dependencies and make KiRi a more pleasant experience to install - and avoid the need for Docker builds too. I found Docker on macOS was really slow and a memory hog. Im told its much better now but id still rather avoid it.

John

leoheck, 2023-01-23:
I am thinking here since Kiri has thousands of dependencies, and since kicad-diff is already made with python. It would be nice to focus on it instead of Kiri. (however, I will try to use kicad-cli on kiri too, to remove extra things since I like how Kiri presents the diff)

Now, for kicad-diff, we could (in this order)
- create a plugin
- check if it can be added to Kicad PCM
- replace plot_kicad_pcb with kicad-cli
- add sch as well
- improve/fix the textual/objects diff
- refactor a bit
- improve the layout of the webpage

Since the plugin is easier, I would like to try to port my plugin to it so we can check how it works with kicad-diff, manually. 

If it works nicely, we can ask if Kicad team can add it to the PCM.

Right now, when I call kiri from kicad, it compares the current state of the sch/board with the previous commit, which is more than enough.

But later, we could add a window asking for the user to select the commits he wants to compare. This is when running with the plunging. 

If running it standalone, we can keep using the current window that selects commits.

leoheck, 2023-01-23:
I am thinking here since Kiri has thousands of dependencies, and since kicad-diff is already made with python. It would be nice to focus on it instead of Kiri. (however I will try to use kicad-cli on kiri too, to remove extra things since I like my webpage there)

Now, for kicad-diff, we could make
- use kicad-cli
- add sch as well
- improve/fix the textual/objects diff
- create a plugin

Since the plugins is easier, I would like to try to port my plugin to it so we can check how it works with kicad-diff.
Right now, when I call kiri from kicad, it compares the current state of the sch/board with the previous commit. 

Kicad-diff can start with that, but we can add later a screen where the user can select the commits he wants to compare.

Gasman2014, 2023-01-23:
Yes, that sounds like a good order of battle. 

> improve/fix the textual/objects diff
With regard to this, I came across this KiCad parser [https://kiutils.readthedocs.io/en/latest/usage/getting-started.html]. I think this could be quite helpful in improving the text diff. What I am really after is a net aware diff - my feature request to tag the svg with the netname was (politely) declined recently.

I think I pointed out ‘Allspice’  [https://www.allspice.io] to you a while back. I really like the way they blend multiple layers with shades of grey and highlight the changes in green/red or yellow. Unless we had a net aware svg, I don’t think we can easily do this but I am experimenting with blending svg files to see what is possible.

John

leoheck, 2023-01-23:
Ah, the plugin I put inside kicad-diff project sometime ago works already. I just forgot about it. We just have to submit it to PCM.

Can you test it @Gasman2014 ?

This is the plugin.
https://github.com/Gasman2014/KiCad-Diff/tree/master/kicad
You can run my script `install_plugin_v6.sh` to install it.
It will install it in the kicad 6, kicad nightly may have a different folder, at least here on linux

In the PCBNew you should see a toolbar icon with this image
[image]

Once you launch it you are going to see its GUI
[image]

leoheck, 2023-01-23:
Ah, it works with Kicad v6 but not with v7 yet, since xWidgets was compiled is using version 3.1. and no 3.2.* another issue to add to the stack. haha