Gasman2014/KiCad-Diff #89: No module named 'dateutil' on Mac M1 with venv

BenRoe, 2023-01-16:
Hello,
i followed the install instructions and installed the mac needed software via brew.
I use venv to encapsulate the requirements.

```bash 
python3 -m venv venv
source venv/bin/activate
python3 -m pip install -r requirements.txt
python3 -m pip list

Package         Version
--------------- -------
dateutils       0.6.12
Pillow          9.4.0
pip             22.3.1
python-dateutil 2.8.1
pytz            2021.1
setuptools      65.5.0
six             1.15.0
```

If i run `kidiff -h` in the KiCad-Diff folder, i get the following error.

```bash
Traceback (most recent call last):
  File "/Users/ben/Git/KiCad-Diff/bin/../kidiff/kidiff.py", line 27, in <module>
    import scms.svn as svn
  File "/Users/ben/Git/KiCad-Diff/kidiff/scms/svn.py", line 8, in <module>
    from dateutil.parser import isoparse
ModuleNotFoundError: No module named 'dateutil'
```

BenRoe, 2023-01-16:
Tried the `plot_kicad_pcb` command.

```bash
❯ plot_kicad_pcb /Users/ben/Git/0xCB-3w6/rev1.0/3w6.kicad_pcb
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this
./src/common/stdpbase.cpp(59): assert ""traits"" failed in Get(): create wxApp before calling this

Kicad (PCBNew API) version (6.0.9-0)
Board version 20211014 (created with Kicad 6)

 # ID Name          Filename
 1  0 F_Cu          /Users/ben/Git/0xCB-3w6/rev1.0/3w6-00-F_Cu.svg
 2 31 B_Cu          /Users/ben/Git/0xCB-3w6/rev1.0/3w6-31-B_Cu.svg
 3 32 B_Adhesive    /Users/ben/Git/0xCB-3w6/rev1.0/3w6-32-B_Adhesive.svg
 4 33 F_Adhesive    /Users/ben/Git/0xCB-3w6/rev1.0/3w6-33-F_Adhesive.svg
 5 34 B_Paste       /Users/ben/Git/0xCB-3w6/rev1.0/3w6-34-B_Paste.svg
 6 35 F_Paste       /Users/ben/Git/0xCB-3w6/rev1.0/3w6-35-F_Paste.svg
 7 36 B_Silkscreen  /Users/ben/Git/0xCB-3w6/rev1.0/3w6-36-B_Silkscreen.svg
 8 37 F_Silkscreen  /Users/ben/Git/0xCB-3w6/rev1.0/3w6-37-F_Silkscreen.svg
 9 38 B_Mask        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-38-B_Mask.svg
10 39 F_Mask        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-39-F_Mask.svg
11 40 User_Drawings /Users/ben/Git/0xCB-3w6/rev1.0/3w6-40-User_Drawings.svg
12 41 User_Comments /Users/ben/Git/0xCB-3w6/rev1.0/3w6-41-User_Comments.svg
13 42 User_Eco1     /Users/ben/Git/0xCB-3w6/rev1.0/3w6-42-User_Eco1.svg
14 43 User_Eco2     /Users/ben/Git/0xCB-3w6/rev1.0/3w6-43-User_Eco2.svg
15 44 Edge_Cuts     /Users/ben/Git/0xCB-3w6/rev1.0/3w6-44-Edge_Cuts.svg
16 45 Margin        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-45-Margin.svg
17 46 B_Courtyard   /Users/ben/Git/0xCB-3w6/rev1.0/3w6-46-B_Courtyard.svg
18 47 F_Courtyard   /Users/ben/Git/0xCB-3w6/rev1.0/3w6-47-F_Courtyard.svg
19 48 B_Fab         /Users/ben/Git/0xCB-3w6/rev1.0/3w6-48-B_Fab.svg
20 49 F_Fab         /Users/ben/Git/0xCB-3w6/rev1.0/3w6-49-F_Fab.svg
21 50 User_1        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-50-User_1.svg
22 51 User_2        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-51-User_2.svg
23 52 User_3        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-52-User_3.svg
24 53 User_4        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-53-User_4.svg
25 54 User_5        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-54-User_5.svg
26 55 User_6        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-55-User_6.svg
27 56 User_7        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-56-User_7.svg
28 57 User_8        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-57-User_8.svg
29 58 User_9        /Users/ben/Git/0xCB-3w6/rev1.0/3w6-58-User_9.svg
``

jnavila, 2023-01-17:
dateutil does not seem to have been ported to python 3.11: https://dateutil.readthedocs.io/en/stable/

BenRoe, 2023-01-17:
> dateutil does not seem to have been ported to python 3.11: https://dateutil.readthedocs.io/en/stable/

Thanks for the hint. Installed Python 3.9.16 with Pyenv, but same problem.

```bash
KiCad-Diff on  master [?] via  v3.9.16 (venv)
❯ kidiff -h
Traceback (most recent call last):
  File "/Users/ben/Git/KiCad-Diff/bin/../kidiff/kidiff.py", line 27, in <module>
    import scms.svn as svn
  File "/Users/ben/Git/KiCad-Diff/kidiff/scms/svn.py", line 8, in <module>
    from dateutil.parser import isoparse
ModuleNotFoundError: No module named 'dateutil'
```

leoheck, 2023-01-19:
@BenRoe try this.

I saw `dateutil` is being used for SVN only. 
Are you using SVN? If not, then try this:

Remove `dateutil` from the `requirements.txt`.

Then, in the file `kidiff/scms/svn.py`  comment out line 8 where the `dateutil` declaration is. 

Like this

``` python
# from dateutil.parser import isoparse
```

Also, replace this, at the end of the file

```python
    def characters(self, content):
        if self.save and len(content):
            if self.save == "date":
                self.current_line += isoparse(content).strftime("%Y-%m-%d %H:%M")
            else:
                self.current_line += content
            self.save = False
```

by this 

```python
    def characters(self, content):
        self.current_line += content
```

Then you will be able to run it. 
I am looking for a way to remove this dependency I just need a way to test it after...

leoheck, 2023-01-19:
I tried to fix this using ChatGPT this time since I am lazy (not really) and since no one uses SVN except by @Gasman2014  hehe (just kidding)

This way I could figure out the format of the content of the log of that line.

[image]

And this way I could change the program removing the useless lib.

[image]

Maybe the for loop is not necessary, but who knows, right?

I am going to make a PR here so you can test if this works for you @BenRoe (if you don't use SVN it will definitely work)

leoheck, 2023-01-19:
here it is, https://github.com/Gasman2014/KiCad-Diff/pull/90
@Gasman2014 do you still use or have access to an SVN repo to test this? It would be awesome.

Gasman2014, 2023-01-19:
I think I have one on an old computer at home. I'll try and spin it up later to check.
SVN is a bit old fashioned but it is a very suitable tool for managing library assets as it enforces file locking and reduces the chances of a merge conflict if two people try to edit an asset.
Thanks for looking at this. As you know, Im a big fan of Fossil SCM.

JP

leoheck, 2023-01-19:
I could setup an SVN repo now and could see it was not working. Then I could fix it. See, it looks like it is working. But I appreciate it if you confirm it too, @Gasman2014, please.

[image]

Gasman2014, 2023-01-19:
Will check later. I'm more than happy to drop the dependency - its only to pretty format a well defined string.
JP

leoheck, 2023-01-19:
I made a mess, so I recreated my PR, https://github.com/Gasman2014/KiCad-Diff/pull/91
I also could test and fix issues when the repo has more than 1 commit

[image]

Gasman2014, 2023-01-23:
Closed via https://github.com/Gasman2014/KiCad-Diff/commit/a055da23a35a1c363f8e037622dbfa29dc0da5eb

closed 2023-01-23 by Gasman2014

leoheck, 2023-01-23:
@BenRoe you can use the version from master branch now. The dateutil dependency is gone. Please, create a new issue if you are having problems with something else.