Gasman2014/KiCad-Diff #92: Consider refactoring to use kicad-cli export

Gasman2014, 2023-01-23:
Consider refactoring the code now that ```kicad-cli``` is available and comes packaged with a KiCad install.. This should make this more portable and not require separate plotting scripts for boards & schematics. Also, it would drop the need for plotgitsch which is a slightly complex dependency. 

Since the kicad-cli is available _outside_ of KiCad, there is no longer any requirements to run any part of the program with the KiCad provided python - which has been a source of complication, particularly on macOS.

```The kicad-cli brings all the necessary functions to automate export elements of schematic and pcb designs.

The currently avaiable commands are:

kicad-cli sch export bom
kicad-cli sch export netlist
kicad-cli sch export pdf
kicad-cli sch export svg
kicad-cli pcb export drill
kicad-cli pcb export dxf
kicad-cli pcb export gerber
kicad-cli pcb export pdf
kicad-cli pcb export pos
kicad-cli pcb export step
kicad-cli pcb export svg
You can use the -h argument to find out more about the command specific options.

kicad-cli sch export svg -h
input: 1 argument(s) expected. 0 provided.
Usage: svg [-h] [--output VAR] [--theme VAR] [--black-and-white] [--exclude-drawing-sheet] [--no-background-color] input

Positional arguments:
  input                  	Input file

Optional arguments:
  -h, --help             	shows help message and exits
  -o, --output           	Output file name [default: ""]
  -t, --theme            	Color theme to use (will default to pcbnew settings) [default: ""]
  --black-and-white      	Black and white only
  --exclude-drawing-sheet	No drawing sheet
  --no-background-color  	Avoid setting a background color (regardless of theme)

kicad-cli pcb export svg -h
input: 1 argument(s) expected. 0 provided.
Usage: svg [-h] [--output VAR] [--layers VAR] [--mirror] [--theme VAR] [--black-and-white] [--page-size-mode VAR] [--exclude-drawing-sheet] input

Positional arguments:
  input                  	Input file

Optional arguments:
  -h, --help             	shows help message and exits
  -o, --output           	Output file name [default: ""]
  -l, --layers           	Comma separated list of untranslated layer names to include such as F.Cu,B.Cu [default: ""]
  -m, --mirror           	Mirror the board (useful for trying to show bottom layers)
  -t, --theme            	Color theme to use (will default to pcbnew settings) [default: ""]
  --black-and-white      	Black and white only
  --page-size-mode       	Set page sizing mode (0 = page with frame and title block, 1 = current page size, 2 = board area only) [default: 0]
  --exclude-drawing-sheet	No drawing sheet

```

This tool would facilitate diffing of net lists and pos files too and comparing step models would also be possible.

leoheck, 2023-01-23:
It is sad they have split kicad in kicad-cli. But having a cli is indeed a victory. Which version has this kicad-cli? Did you already try it?

Gasman2014, 2023-01-23:
Hi Leandro,

Kicad-cli been in the nightly development version ‘6.99’ for a while now and seems to work OK. There is already a KiCad 7.0-RC apparently so I expect the proper 7.0 is only a few weeks away. 

Best Wishes

John

Command Line Interface
Mark Roszko <https://gitlab.com/mroszko> in a slew of commits has added a new kicad-cli binary to the KiCad packages.

 <https://www.kicad.org/blog/2022/12/Development-Highlight-December-Edition-features-coming-to-KiCad-7/cli.png>
The separate kicad-cli binary was necessary to work around limitations on applications being both GUI and console on Windows at the executable level. Along with wxWidgets having a distinction between a GUI and console app for initializatin.

The kicad-cli brings all the necessary functions to automate export elements of schematic and pcb designs.

The currently avaiable commands are:

kicad-cli sch export bom
kicad-cli sch export netlist
kicad-cli sch export pdf
kicad-cli sch export svg
kicad-cli pcb export drill
kicad-cli pcb export dxf
kicad-cli pcb export gerber
kicad-cli pcb export pdf
kicad-cli pcb export pos
kicad-cli pcb export step
kicad-cli pcb export svg
You can use the -h argument to find out more about the command specific options.

These changes have also enabled the STEP export to work headless on Linux, unlike the original kicad2step binary.

We hope this is a useful mechanism for users to adopt. We do plan to introduce a GUI based jobs system in KiCad 8 but see this as a interim solution.

leoheck, 2023-01-23:
Yeah, I could find it. 

I could export a good black and white svg which is exacly what kicad-diff likes
```
kicad-cli-nightly sch export svg --black-and-white --no-background-color board.kicad_sch
```

leoheck, 2023-01-23:
Pages and layers have to be exported individually manually.

```
# You have to export each page, individually..
kicad-cli sch export svg --black-and-white --no-background-color board.kicad_sch

# You have to export each layer, individually...
kicad-cli pcb export svg --layers "F.Cu" --black-and-white board.kicad_pcb
```

I could not see any way to export all at once using kicad_pro

leoheck, 2023-01-23:
Unfortunately, it does not export .sch files, it could support this since it already has this feature inside the code.

```
2S_serial_charging_board …
➜ kicad-cli-nightly sch export svg --black-and-white --no-background-color board.sch      
Failed to load schematic file
```

Gasman2014, 2023-01-23:
Thought the -l switch took a comma seperated list of layers? Not by my computer now.

John Pateman

Gasman2014, 2023-01-23:
Worked for me - able to export svgs from the schematic - although I didn’t try the no-background or black-and-white switches. How old is your nightly? And I only needed “kicad-cli” not kicad-cli-nightly.

John Pateman

leoheck, 2023-01-23:
My is the lates niglty. The names may differ because you are on MacOS and I on Linux. 

Now, thinking here, kicad-cli for PCB is not better than what kicad-diff has except that it gives us the schematics too.

leoheck, 2023-01-23:
For instance, we already have to iterate through the layer names to export each one individually. I could not find a way to export all the existing layers at once. (I did not try hard, tho)

For the schematics, you can pass the main one and it is going to export all the available pages.

leoheck, 2023-01-24:
@Gasman2014 I give it a quick try to plot schematics using kicad-cli

Here is my working branch of Kidiff.
https://github.com/leoheck/KiCad-Diff/tree/kicad7

It is currently working with projects that have a single schematic page. 
Also, the project has to be Kicad >= 6 (does not work with .sch files)

This is a taste of what I have working...

Gasman2014, 2023-01-24:
kicad-cli pcb export svg -l F.Cu,F.Fab,F.SilkS,F.Mask,Edge.Cuts TEST.kicad_pcb

This produces a composite image of several layers at once. I think we could helpfully combine some layers to cut down on the number of layers shown - I would like to shade some of these in grey

But you are right, we are currently iterating through the layers and if we currently go with the same layout we will just need to iterate this with different layers, just as we do at the moment.

Can’t seem to get the ‘theme' switch working though - I’m going to report it.

leoheck, 2023-01-30:
> This produces a composite image of several layers at once. I think we could helpfully combine some layers to cut down on the number of layers shown - I would like to shade some of these in grey

This could give us extra work to make the diff and it may be hard for me since the job is in javascript in the SVG Pan and Zoom lib. 

I was considering using kicad-cli on my Kiri tool too, but it looks like I won't gain much doing that since it does not support Kicad 5 (*.sch) schematics. And it is important since I am allowing a review of the git history. Also, as I could see on the Kicad forum, the developer looks a bit pissed with this work to believe it would be possible anywhere soon, haha.

leoheck, 2023-01-30:
Did you try my branch `kicad7` above? It shows that it is possible to add schematic diff, easily, to Kicad-Diff using Kicad 7.

Gasman2014, 2023-01-30:
I did try the branch last week but ran into problems with the ‘wx framework build required’. I did eventually get it running under ‘pythonw’ but hit a couple of other snags and haven’t had a further chance to play.I think the developer is keen to support KiCad-cli and the code made the feature freeze in September but because of the string freeze prior to the release of 7, they can’t do much more at this stage until 8 is in development. I have posted a bug about the themes, for instance and it does seem to be being looked at positively.Obviously some of our needs are a bit niche and need a bit of upvoting in gitlab to gain traction. My request to tag the svg elements with the net name would have been useful but has been rejected, for instance. I had hoped that KiCad-cli would provide a platform agnostic way of generating the images without all the faffing around with passing off the KiCad bundled python - I think it will come but maybe we’ll have to wait for 8 for a more complete solution.Sent from my iPad

leoheck, 2023-01-31:
I agree. I pushed some changes to the same branch if you would like to test it again. Now it plots all schematics pages if there are extra pages in the project.

[image]

leoheck, 2023-02-17:
This script exports all schematic pages and each layer of the PCB to a unique file in case someone is interested

https://gist.github.com/leoheck/0023e5118fb537147f045a2df7af1c3e

leoheck, 2023-02-17:
Things I saw while using this kicad-cli v7

- It does not plot schematics from Kicad v5 (those that use .sch extension)
- Plotting each layer of a PCB individually is way slower than using the current Kicad-Diff  plot_kicad_pcb script to do the same task.

leoheck, 2023-02-17:
Also, if you just have the Kicad v7 `plot_kicad_pcb` does not work anymore. I am not sure if it is missing python libraries or if the path has changed.

leoheck, 2023-02-20:
I forgot to mention here, but I made a PR fixing Kicad-Diff it for Kicad v7
I still have to validate on MacOS. I will be happy if someone is interested in verifying that.
https://github.com/Gasman2014/KiCad-Diff/pull/94

Cheers.

leoheck, 2023-02-21:
It was validated on a M1 too. It is all clear.

leoheck, 2023-02-21:
I am going to close this issue now that I merged the PR to support Kicad v7. We can reopen it in the future or start a new one if needed. Cheers!