Gasman2014/KiCad-Diff #93: Error: wxPython installing on Mac M1

BenRoe, 2023-01-24:
Hello,
i get an error while installing `wxPython`.

- Mac M1 13.1
- 3.9.16 Python via Venv 
- Master branch from 23.01

How to reproduce:
```bash
cd /Users/foobar/git/KiCad-Diff
python3 -m venv .venv
source .venv/bin/activate
export CFLAGS=-I/$(brew --prefix)/include
export CXXFLAGS=-I/$(brew --prefix)/include
python3 -m pip install -r requirements.txt
```

Error:
```bash
     ...
     fatal error: too many errors emitted, stopping now [-ferror-limit=]
      20 errors generated.

      Waf: Leaving directory `/private/var/folders/p0/dj72mqbx01qcw5779ycqh1140000gn/T/pip-install-wcic0by8/wxpython_f5181bfafeb5453ea9b44afecb4fc2e0/build/waf/3.9'
      Build failed
       -> task in '_webkit' failed with exit status 1 (run with -v to display more information)
       -> task in '_webkit' failed with exit status 1 (run with -v to display more information)
       -> task in '_webkit' failed with exit status 1 (run with -v to display more information)
       -> task in '_webkit' failed with exit status 1 (run with -v to display more information)
       -> task in '_webkit' failed with exit status 1 (run with -v to display more information)
      Command '"/Users/ben/Git/KiCad-Diff/.venv/bin/python3" /private/var/folders/p0/dj72mqbx01qcw5779ycqh1140000gn/T/pip-install-wcic0by8/wxpython_f5181bfafeb5453ea9b44afecb4fc2e0/bin/waf-2.0.8 --wx_config=/private/var/folders/p0/dj72mqbx01qcw5779ycqh1140000gn/T/pip-install-wcic0by8/wxpython_f5181bfafeb5453ea9b44afecb4fc2e0/build/wxbld/wx-config --python="/Users/ben/Git/KiCad-Diff/.venv/bin/python3" --out=build/waf/3.9 configure build ' failed with exit code 1.
      Finished command: build_py (1m37.453s)
      Finished command: build (3m5.547s)
      Command '"/Users/ben/Git/KiCad-Diff/.venv/bin/python3" -u build.py build' failed with exit code 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> wxPython

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
WARNING: You are using pip version 22.0.4; however, version 22.3.1 is available.
You should consider upgrading via the '/Users/ben/Git/KiCad-Diff/.venv/bin/python3 -m pip install --upgrade pip' command.
```

Current Solution:
```bash
pip install --pre -f https://wxpython.org/Phoenix/snapshot-builds wxPython
# or if you use venv
python3 -m pip install --pre -f https://wxpython.org/Phoenix/snapshot-builds wxPython
```

leoheck, 2023-01-24:
Is the issue solved? If yes, then, please, close it.

BenRoe, 2023-01-24:
> Is the issue solved? If yes, then, please, close it.

If i follow the instructions from the [readme](https://github.com/Gasman2014/KiCad-Diff/blob/master/README.md#instructions) i get the error. 
I don't know if i should add my solution to the readme, because it's not a "official" solution.

leoheck, 2023-01-24:
There is no official solution. You can solve the depencies the way you prefer or the way it is better for your system. You are already doing that with virtual environment, which is a good thing.

closed 2023-01-24 by leoheck

Gasman2014, 2023-01-25:
FWIW, it seems to install cleanly with `conda` which is what I use for virtual environment management on macOS.