Gasman2014/KiCad-Diff #95: Errors while generating

chuckwagoncomputing, 2023-03-11:
Recently we started getting errors. This happens with both Kicad 6 and Kicad 7.
We're using @leoheck's branch.
```
Traceback (most recent call last):
  File "/home/david/Downloads/rusefi/KiCad-Diff/bin/../kidiff/kidiff.py", line 934, in <module>
    commit1, commit2, commit_datetimes = scm.get_boards(kicad_pcb_path, repo_path, kicad_project_dir, board_filename, commit1, commit2)
TypeError: cannot unpack non-iterable NoneType object
```
Most recent run without errors was 2/12/23: https://github.com/rusefi/rusefi/actions/runs/4158982349
First run with errors was 2/25/23: https://github.com/rusefi/rusefi/actions/runs/4270675246

leoheck, 2023-03-11:
Ah, try to use the version of the master of this repo this time. I think my version has diverged a bit, but the master on [Gasman2014](https://github.com/Gasman2014)[KiCad-Diff](https://github.com/Gasman2014/KiCad-Diff) may have the latest fixes.

leoheck, 2023-03-11:
My branch may have issues since I am looking for the best way it will be for Kicad 7.

chuckwagoncomputing, 2023-03-12:
I checked just now and this repo has the same behavior.

chuckwagoncomputing, 2023-03-12:
Also, in my checking, I tried older versions of your repo, and it still behaved the same with 3205daaf282bea2ae0a6763545cae4cffaac7b1e, which was before the last error-free run, so something else must have changed, not your code.
I've done a bit of digging, but haven't found the culprit yet.
The two jobs were on different Github runner images.
[Working run image info](https://github.com/actions/runner-images/blob/ubuntu22/20230206.1/images/linux/Ubuntu2204-Readme.md)
[Broken run image info](https://github.com/actions/runner-images/blob/ubuntu22/20230219.1/images/linux/Ubuntu2204-Readme.md)

The only thing I've tried the older version of so far is Git, but that didn't fix it. (Though I tried that on my laptop, which is running Arch rather than Ubuntu)

chuckwagoncomputing, 2023-03-12:
Also, it only gives the error when there weren't any changes between the two commits. If there were changes, a diff is produced without error.

chuckwagoncomputing, 2023-03-12:
Possibly because get_boards returns nothing in https://github.com/Gasman2014/KiCad-Diff/blob/b3052aeadb3b78e609b91f964ad391d1b17eb105/kidiff/scms/git.py#L47
Though that line was added Jan. 30, so I don't understand how the Feb. 12 run didn't give any errors.