Gasman2014/KiCad-Diff #98: KiCad-Diff for windows KiCAD users.

Manuel-Baez-Ponce, 2023-12-27:
Hello, dear KiCad-Diff Developers!

I use the kicad in Windows 10 and in the project Readme file, I haven't found the Windows installation section only the Linux and MacOS sections, so I wanna ask you if this plugin isn't available for Windows, on other hand, I am very interested in this functionality and I would like to contribute to this project, I am a beginner on git and kicad, but still, I was wondering if you could instruct me on how the code works and its documentation, as well as how I can start collaborating if it's not too much trouble.

Gasman2014, 2023-12-27:
I’ve not been able to maintain this properly for a while, unfortunately and I don’t have a windows machine set up either. As far as KiCad Diff is concerned, it is Python with a few modules so should be pretty platform agnostic.It works by allowing a user to select two versions to compare in a file selector viewer. It then generates the relevant KiCad layout files. The layout files are processed into svg files. Each layer pair is compared using a feMatrix transform which keeps common elements grey and additions/removals are either cyan or magenta.You might want to check out KiRi https://github.com/leoheck/kiriThis uses my original code but has extended it to include schematic diffing and has a neat way of looking at comparisons between any arbitrary pair of revisions.Since these were written, KiCad-cli has become available which might simplify some of the code to generate the necessary layers.I could see how this could be improved - I think that multiple layers of front/back or inner layer sets could have different opacity settings and the project needs a bit of a polish.Best WishesJohnSent from my iPad

leoheck, 2023-12-27:
Yeah, I would say that if you have WLS installed, you can follow Linux instructions.

leoheck, 2024-07-02:
I am closing this issue since the author of this issue did not respond. It can also be installed easily on windows using WLS.